Ultimaker/Cura #9479: Exceptions for Post Processing Script Warning Indicator

SofaKingRotten, 2021-03-30:
### Is your feature request related to a problem?

I love the post processing warning but because I always use the LCD display script I always see the warning. This does not help warn me when it is always turned on while using (default) and other useful scripts.

### Describe the solution you'd like

By adding a tickbox in post processing scripts list, like "Don't warn me about this script". Now I will only see the warning indicator when I run any other script like "Pause at height", which I use frequently, but need a warning to shut it off for the next object.

### Describe alternatives you've considered

I could shut off the LCD display script but, heck, I kinda like it.

### Affected users and/or printers

Anyone who uses SOME scripts all the time, but then use a few other scripts SOME of the time. I think everyone using the default "LCD Display" is affected.

fvrmr, 2021-04-02:
Hi @SofaKingRotten thank you for your feature request. 
You are mentioning a warning indicator and I think you mean the red dot, which is more an indicator. To let you know that you are using post processing scripts. Maybe the color red isn't the right color to indicate that. I will discuss that with our UX designer. 

I believe removing that for only the LCD display is not a solution. Since you can see how many post processing scripts are there. 
[image]

Let me know what  you think!

SofaKingRotten, 2021-04-03:
The problem is, everybody uses the default “Display filename and layer on screen” in Cura. That means, that warning indicator is ALWAYS showing up. You kinda just get used to seeing it, even though you know it is only because of that LCD Display script that is always enabled. Until you enable another script like “Pause at height”, then forget to disable it later because you forgot to shut it off, because you always see the warning on and didn’t notice it. Then your next print hangs for hours waiting for you…
 
My thought was, every script in the list has a tickbox next to it. If you don’t want to be warned about leaving THAT particular script enabled, you can check the tickbox. The warning indicator would only show up id there was at least one script without the tickbox checked. Now whenever I need to “Pause at height”,  I would notice the warning easier since I don’t see it every day/hour.
 
Let’s fix it-Curt

 

 <https://user-images.githubusercontent.com/77000924/113414455-53dc9880-93bd-11eb-84de-bfca88f1b761.png> Image removed by sender. Screenshot (151)

fvrmr, 2021-04-16:
Hi @SofaKingRotten we have discussed this issue with the team and we decided to defer this issue because this could confuse other users. Hope you understand. 

Also I discussed the notifications with our UX designer and we want to improve the design of that button to integrate it better in the Cura UI in the future.

SofaKingRotten, 2021-04-16:
Well, if they REALLY understood what I am asking, I believe  it is less confusing then seeing the warning indicator BEAMING at me on every job, even if I only have the LCD Display script running. The warning seems useless if it is always on. Somebody else out there MUST have the same thoughts on this, but for now, I digress...

closed 2023-07-13 by github-actions[bot]