Ultimaker/Cura #9487: Cura produces non-smooth Z-value transition in vase mode

BarsMonster, 2021-04-01:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ghost4

### Reproduction steps

1. Slice Xtest-HR.stl in vase mode (Spiralize outer contour checked)
2. Every whole millimeter of Z-height Z coordinate will jump by 0.002mm instead of uniform 0.001mm changes. 

### Actual results

This non-smooth Z coordinate change creates visible defects, especially in low microstepping mode (microstepping 2, T8x2mm lead, 0.9° stepper). In some of these jumps (not all) printer makes 2 loops with exactly same stepper position, which creates visible surface defect. 

### Expected results

Smooth Z change, each next Z coordinate is exactly 0.001mm larger than previous. 

Archive includes: Project file, STL used for slicing, exaple of gcode. Python script to analyze gcode. Excel file to show the graph:

[Cura-report.zip](https://github.com/Ultimaker/Cura/files/6241086/Cura-report.zip)

Graph is made in Excel, in logarithmic Y-axis:

[image]

jellespijker, 2021-04-02:
Hi @BarsMonster Great analysis, unfortunately I'm swamped for today, but I will look into it on next Tuesday and come back to you.

GregValiant, 2021-04-02:
That is an excellent analysis.  Unfortunately, the problem appears to be in your settings, and the printer itself may be suffering from mechanical Z-banding.
I printed your gcode file without making any changes.  I then I sliced the STL file using my profile with Spiralize enabled (with no changes to anything else) and printed the subsequent gcode.  The printer is a well maintained Ender 3 Pro.
Within your settings I noted that Accel and Jerk were not enabled.  The brim went down OK but then the print head motion in your gcode file was extremely herky-jerky.  This resulted in numerous blobs around the print likely do to either buffering problems in the printer/processor or a lack of Accel and Jerk control.  I did not get the Z stripes (shown in your image) and so I believe they are evidence of a mechanical issue with your printer.  It might be as simple as loose set screws on the Z coupler.
You will notice that neither print shows Z-banding.

I will leave it up to you to figure out which print was from your gcode.

[image]

BarsMonster, 2021-04-02:
@GregValiant Thanks for trying it! 

1) This gcode is sliced with extremely fine resolution which will definitely cause stutter on 8-bit boards. I am printing on SKR-PRO. So I agree that printing this gcode on 8-bit machines would not reproduce the issue. I think defects that I am mentioning are quite weak and could definitely be masked by stutter noise. 

2) These Z stripes are definitely not mechanical issue. With 2 microstepping I can hear every step on Z axis (X & Y are silent). On these bands Z axis stops "ticking up". So it prints whole cycle with the same Z coordinate. This causes some slight over-extrusion. This lack of "ticking" on certain heights made me analyze progression of Z coordinates. 

3) I noticed that this effects depends on Z-axis resolution. When I increase micro-stepping (with relevant change of steps per mm), from full step to 2, then to 16 - these bands become less and less visible. That is why suspect it is relevant to rounding / luck. So many printers with 1/16 microstepping might see very slight defects which are harder to notice (but still imperfect). I started investigation with much weakers defects, until I found that larger steps make it more visible.

GregValiant, 2021-04-02:
It sounds to me like it's more a question of coming up with a settings profile that works with your printer.
Unless your Z system is absolutely frictionless - asking for a movement that provides .001mm steps of vertical motion can get lost in the pieces parts.  Trying to push an assembly to give higher tolerance than what was used to manufacture the individual parts of the assembly would seem to be an exercise in futility.
Regarding my test prints, yes it's a stock 8 bit Creality mainboard but by holding the maximum resolution at 0.4 and max deviation at .025 the number of very short line segments that make up the arcs is held way down and there are less lines of gcode to process.  I keep the Accel and Jerk at a conservative 500 and 8 respectively.  The end result is smooth consistent motion.  You mentioned that you have a 32 bit board.  From what I gather, the main advantage isn't processing speed but rather that the 32 bit boards come with more memory and so more firmware options can be un-leashed for use.  Additionally, we're talking about plastic here.  A .001mm step is really unnoticeable.  Blobs on the other hand are butt ugly.
I think you need to balance the resolution of the slice to what the machine is actually capable of.  It took me a bit, but mine is essentially a stock printer and I just don't ask it to do things that (I now know) it can't.  I think if you go back to one of the standard profiles and keep tweaking the slicing resolution finer and finer that you will come to a point where the machine can no longer keep up.  Move the settings back a notch and you are there.  At some point the molten plastic is going to have it's own limit concerning resolution.
Diving into all those numbers is interesting but don't forget to read the finished print because at the end of the day, it's the only thing that's important.

BarsMonster, 2021-04-02:
I've compared spiralized vs regular (non-vase) versions. 

Regular print is perfect. G-code shows rock-solid 0.200 steps as expected. Which suggests that it is not mechanical issue with the printer. 

[image]

GregValiant, 2021-04-02:
Your Vase mode print looks perfect until it gets about 6mm up.  The spacing of the ridges above that look pretty darn regular.
The three I printed were all vase mode and are 20.00 +/-.03.  Certainly acceptable.  If I put them all out in the Florida sun they would be +.5 or something.

Ghostkeeper, 2021-04-15:
Slicing your project file in Cura, I don't see the result in the g-code either. Here is a snippet of g-code from the project file produced by `Xtest-HR.3mf`:

```
;LAYER:2
;TYPE:WALL-OUTER
;MESH:Xtest-HR.stl
G1 F3000 X89.816 Y67.731 E106.84017
G1 X89.995 Y67.486 E106.85636
G1 X90.182 Y67.242 E106.87277
G1 X90.377 Y67.002 E106.88928
G1 X90.577 Y66.767 Z0.561 E106.90575
G1 X90.788 Y66.531 E106.92265
G1 X90.979 Y66.326 E106.93761
G1 X91.209 Y66.092 E106.95513
G1 X91.437 Y65.872 E106.97204
G1 X91.66 Y65.668 E106.98817
G1 X91.89 Y65.466 Z0.562 E107.00451
G1 X92.112 Y65.283 E107.01987
G1 X92.377 Y65.074 E107.03789
G1 X92.607 Y64.901 E107.05325
G1 X92.852 Y64.727 E107.06929
G1 X93.114 Y64.55 E107.08617
G1 X93.368 Y64.388 Z0.563 E107.10226
```

So I'm seeing it increment to `Z0.0561`, `Z0.0562`, `Z0.0563`, etc. without any jumps. Similarly in your own .gcode file in the zip archive.

Inspecting the `g-stat.py` file reveals why you might be seeing some 0.002mm increments: It's a bug in the Python script. The Python script finds parts of commands matching with the regex `^.*Z(\d+\.\d+).*$`. This matches on every Z coordinate consisting of `###.###`, with at least one number before and after the period. However not all Z coordinates are formatted this way. For instance, in lines 16570 through 16582, it goes to `Z1.999`, `Z2`, `Z2.001`. The move to `Z2` would not get picked up by your Python script, resulting in that your spreadsheet will just have `Z1.999` and `Z2.001`, which would get picked up as a move of 0.002mm indeed. But there's really another command in between.

Try it with a more robust regex: `Z(-?\d+(\.\d+)?)`.

So far I'm unable to reproduce your issue. You might indeed just be seeing some stick in the Z stage.

BarsMonster, 2021-05-01:
I confirm that issue was in my code, and CURA is innocent here :-)

closed 2021-05-01 by BarsMonster