Ultimaker/Cura #9489: Cura Crashes all the time, tested with 3 versions

13cos13, 2021-04-01:
### Application Version

4.8/4.7/4.6

### Platform

windows 10 pro 64

### Printer

Alfawise u20

### Reproduction steps

Just launch, then it close

### Actual results

Sometime i have time to make some moves inside, but most of the time i dont have time to do anything because it closes instantly

### Expected results

i would love it to start..

I also tested all solutions topics from google with words "cura crash 4.8"
this is a new computer but drivers are up to date, and directx too.
Ryzen 3800x, Asus primes x570, sapphire radeon 5700xt
[cura.log](https://github.com/Ultimaker/Cura/files/6244510/cura.log)
plz i need your help, i have no clue of what i can do to solve this..

DrGoldbond, 2021-04-02:
HARD CRASHING and will not stay open longer than .5 seconds.

I have tried many different versions of Cura. I do have all the updates from Microsoft Installed. I do have all the updated 5700 XT drivers for my graphics card installed. I have 32GB of DDR4 Ram, over 4TB of hard drive space, and NOTHING else running after a fresh restart of my system. Something changed between last year and this year and I'm now getting this force close or system freeze issue and it was not my hardware. I used to be able to run this just fine. My $2000 computer has no issue handling this application, something in the Windows 10 software or Cura software is the cause. Looking around online and many people have the same issue with 0 working fixes for it.

fvrmr, 2021-04-02:
Hi @13cos13 thank you for your report.
We are aware of an issue where Cura crashes on certain machines.
[Here](https://support.ultimaker.com/hc/en-us/articles/360018633500-Ultimaker-Cura-won-t-run-on-my-Ryzen-Graphics-card?source=answerbot&auth_token=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJhY2NvdW50X2lkIjo5NTU2NjUwLCJ1c2VyX2lkIjozNzc1MTI2NzE1MTksInRpY2tldF9pZCI6NTc3NywiZGVmbGVjdGlvbl9pZCI6MzYwMTYxMDgzNjk5LCJhcnRpY2xlcyI6WzM2MDAxODYzMzUwMCwzNjAwMTE0MjczNDAsMzYwMDE2NTM4MDAwXSwidG9rZW4iOm51bGwsImV4cCI6MTYxMjQ0MDc3Nn0.1QtS8bkpGcpMqyLVdXxP78ReAKU0Uo0a8sqPa9WVTc4) you can find more information and a work around. 

Let me know if this works for you.

13cos13, 2021-04-02:
Hi fvrmr,
i already tested this method. this didnt work.
actually i have tryed ;
- Modify cfg
- Delete cfg
- delete all appdata\cura\ folder
- run as admin/compatibility win8/fullscreen opti
- versions 4.8 but also 7 and 6.
- force card use in windows graphics
- same in radeon graphics
- simple screen

none of these solutions worked, but, on 50 trys maby, i can start it.. and when it start and dont crash in 1 min, i can use it normally or quite. it seems have some litle freezes on actions..

13cos13, 2021-04-04:
I just tested Arachnee. Same problem.. T.T

13cos13, 2021-04-04:
same problem on prusa slicer... omg
i can start it but crash after some moves.
What could be wrong ? i can use Maya, All Adobe, 3D games, cryptomining, but i cant use any slicer at the moment...

13cos13, 2021-04-11:
New windows, same problem but worked fine while 3 days...
i also tested 4.9 but there is always a crash with QT5something.dll, on every version.
sometimes i can start it, and works fine until next start.. 1 time on maby 30-50 trys.
i really need help plz

Ghostkeeper, 2021-04-15:
The link provided is specifically for a crash between Ryzen and ASUS motherboards that causes the entire computer to lock up, not just Cura to crash. Since the symptom is completely different I think that's not the solution you're looking for.

There is a recent bug report with the same symptom as you. This guy seems to have found a workaround: https://github.com/Ultimaker/Cura/issues/9456#issuecomment-809704449
Could you try switching the graphics card to graphics mode as per these instrutions? https://www.amd.com/en/support/kb/faq/dh-024

13cos13, 2021-04-18:
Thank u but that didnt help.
The new radeon software dont have this option. But you got the same option in win 10 advanced graphics. I did already tested both with no results..(testing all parameters in Rad Software 1 by 1)
Actually im going back(or trying) to win 7. But i missed drivers injection for ssd M2. I will try on classic in fist time to see if it solved the problem..

13cos13, 2021-04-21:
I cant go back to win 7... /cry

13cos13, 2021-04-25:
Big Thx for no answers ! 

I found how to reproduce this bug, i just made it on an old cpu too.
Cura + BetterHash + Win 10 64 = Problem.
(the problem dont start immediatly, but the second day... still dont understand why)

Ghostkeeper, 2021-04-29:
> Big Thx for no answers !

You're welcome to get an answer, but not to beg for one.

> Plz can you tell me all that is linked to theses QT5 files ?

The PyQt5 is linked to Qt. It provides Python bindings for Qt. We build Qt according to the instructions written in the cura-build-environment repository: https://github.com/Ultimaker/cura-build-environment/blob/master/projects/qt.cmake . And PyQt on Windows is obtained via Pip: https://github.com/Ultimaker/cura-build-environment/blob/master/projects/pyqt.cmake

I wonder how BetterHash could cause Cura to crash? The operating system should separate the processes, right? Is BetterHash doing some hacking outside of their own RAM territory?

13cos13, 2021-04-29:
Sorry for beg, but im trying to solve a problem since 20+ days long.. immagine how many installs, virtual machines, hours of analyses... im getting mad how this is killing my activity.. sorry.
i dont understand, like you, what is the link between cura and betterhash. but i have reproduced this crash on a second computer with different config, and with only windows 10 64, drivers install, cura and better hash. 
I sended a mail to betterhash team too as you can imagine. But actually no answear.
I try to understand the links you gave me (thank you) but as im not developer i dont know how i can use this.
Im not able to answear your last question sorry, im not even sure to understand it. But for the processes, i tooks some hours long making some tests by killing manually some processes that could be in conflicts. I did have a look for starting process on windows opening. But all theses trys didnt solve.
Only slicers softwares are impacted. I can use Maya, all Adobe, recents games, mining... all without problems. But slicers softwares seem impacted by freezes causing crash(sometimes there is a driver alert "time out"). All the time its specified crash by a QT file, but not always the same. And this is only after use of betterhash, even with reinstall.
I hope my description can help you have a suggestion for me.. this is probably something too high for my competances and i dont know how to have some help. Any suggestions welcome.

Ghostkeeper, 2021-05-05:
You could try installing some other software that runs on Qt, such as [OpenSCAD](https://openscad.org/) to see whether the problem is localised to Qt or not.

13cos13, 2021-05-05:
Openscad closing same way ; no alert, just close itself. It look like it close itself after some time and not after some actions from me.
Nom de l’application défaillante openscad.exe, version : 0.0.0.0, horodatage : 0x60172571
Nom du module défaillant : ntdll.dll, version : 10.0.19041.610, horodatage : 0xe5d7ed5c
Code d’exception : 0xc0000374
Décalage d’erreur : 0x00000000000fed29
ID du processus défaillant : 0x938
Heure de début de l’application défaillante : 0x01d741db526a95d4
Chemin d’accès de l’application défaillante : C:\Program Files\OpenSCAD\openscad.exe
Chemin d’accès du module défaillant: C:\Windows\SYSTEM32\ntdll.dll
ID de rapport : ba2dd92f-17d4-4a2d-8bc3-e66c49bfd4ec

Ghostkeeper, 2021-05-07:
I fear then that you have a bug in Qt on your hand, and it won't be something we can likely fix on our side.

The error code `0xc0000374` it gives is a heap corruption. This could be the result of running out of memory. Maybe BetterHash is just taking so much RAM that there is nothing left for Cura to take? That could also explain why it only happens after a certain time.

Cura generally needs ~500MB of free RAM to be able to slice normally. For bigger objects it's going to need more though.

I think it's worth your time to monitor RAM usage when Cura crashes.

Ghostkeeper, 2021-05-12:
Cura ships with its own Qt version, so if you were to replace those Qt DLLs you would still have the problem with other Qt-based applications.

Replacing the Qt DLLs would likely also require re-compiling PyQt, since they then need to be linked to different DLLs. It's not easy to compile PyQt.

There is a newer version of Qt 5.15 than the one we are using. We're using 5.15.2, but there is now a 5.15.3 as well. The 5.15.3 is only available for commercial license holders though (which Ultimaker is, but you won't be able to try it yourself). And even the list of bugfixes is not publicly available...

13cos13, 2021-05-12:
Thank you for précisions.
At this point im not able to solve the problem. Im not developper and im quite lost on what you wrote..
I contacted Betterhash team and the guy told me that the software does not have an impact on QT, and no dll install.
But i can reproduce this bug with only 2 softwares installed.. do you think im wrong to still think about an issue because of this software ? its really suspicious to me..
If you have a suggestion on a slicer not using QT maby .. i think im gona try that way.

Ghostkeeper, 2021-05-13:
> But i can reproduce this bug with only 2 softwares installed.. do you think im wrong to still think about an issue because of this software ?

You are right to think that it's probably got to do either with BetterHash or with Qt, or somewhere in between. Neither of them are things we can really do anything about though.

13cos13, 2021-05-17:
Ok, thank you for your time and your help.

closed 2021-05-17 by 13cos13