Ultimaker/Cura #9492: How to properly install plugins for my printer in for CURA 4.8 and OS Ubuntu 18.04?

MaratZakirov, 2021-04-02:
### Application Version

4.8

### Platform

Ubuntu 18.04

### Printer

Zonestar Z6fb

### Reproduction steps

1. Install CURA 4.8 from Ubuntu software center
2. Download resources file from https://www.jianguoyun.com/p/DZIXy4UQyoP1BxjLstUD ZONESTAR_Cura_Resources.zip which contains plugins and resources and readme.txt file with 
<quote>
1. Open the directory where you installed the Cura software, such as "C:\Program Files\Ultimaker Cura 4.7"
2. copy the files in "resources" to "C:\Program Files\Ultimaker Cura 4.7\resources" 
3. copy the files in "plugins" to "C:\Program Files\Ultimaker Cura 4.7\plugins" 

If you use other operating systems, the operation is almost the same
</quote>

### Actual results

I could not find an analog directory in my OS for installig these plugins/resources for my printer

### Expected results

I can find my printer in ADD Printer settings

MaratZakirov, 2021-04-02:
I made AppImage of CURA version work but still why Ubuntu Software version does not work?

fvrmr, 2021-04-02:
Hi @MaratZakirov thank you for your report. 
The reason why is because we only maintain the AppImage of Cura. 
I will close this issue since the AppImage is working. But feel free to reach out if you have more questions.

closed 2021-04-02 by fvrmr