Ultimaker/Cura #9494: Modifier Creates irregular pattern.

omgsh, 2021-04-02:
### Application Version

4.8.0

### Platform

Catalina 10.15.7

### Printer

Ender 3 V2

### Reproduction steps

225w X 225d X 40h model
Placing modifier halfway up the model
Slicing
Repositioning the modifier by dragging
Slicing again

### Actual results

These weird patterns are slicing BELOW a square modifier block that disappear if the modifier is removed.

### Expected results

only the area being modified should have changes.

Another user commented in the FB Group:
Yup, having that issue with shells and modifier blocks, where Cura decides the shells should somehow end and not continue over the boundary despite the modifier only changing infill. But even if i wanted to change (say increase from 4 to 8) # of shells, the behavior should be that 4 shells continue across the modifier boundary and along the outside of the shape while the extra 4 loop around the modifier boundary.

My Reply:
So I found that this was happening after adding the modifier, slicing and then trying to drag to reposition the modifier. While attempting to use the coordinates to exactly place the modifier before attempting a slice, the issue did not happen after slicing or repositioning and slicing again when moving the modifier using coordinates.

I am unable to upload the model

[image]
[image]
[image]

fvrmr, 2021-04-02:
Hi @omgsh thank you for your bug report.
Can you share your project file (file - save project) as a .zip in a comment here on Github?
Than I can have a closer look.

omgsh, 2021-04-02:
I am unable to upload the model, meaning I can't share the project I was working on.

fvrmr, 2021-04-12:
> I am unable to upload the model, meaning I can't share the project I was working on.

You can't upload it on Github? Then you could try sharing it via Google Doc or WeTransfer and share the link here.

omgsh, 2021-04-12:
You are still not getting it. I know how but I will not share the file. Luckily, it’s something you can reproduce as I mentioned that other users have the same issue.

Ghostkeeper, 2021-04-15:
There are several things going on here. I'll first try to follow your reproduction steps as good as I can.

> 225w X 225d X 40h model

I made a model that has a similar structure to what's shown in the problematic area of your screenshot.
[image]

> Placing modifier halfway up the model

I'm assuming you meant a cutting mesh here, like shown in the last screenshot.
[image]

> Slicing

Layer view looks like this then:
[image]

There is nothing _below_ the cutting mesh. It did cut up the mesh into two pieces, one piece within the cutting mesh and one piece outside of it.

> Repositioning the modifier by dragging

[image]

> Slicing again

The result is similar as the first time I sliced. Just in a different spot.
[image]

> only the area being modified should have changes.

This is not quite the expected behaviour of the cutting mesh. The expected behaviour is that it cuts the model into separate pieces, which can then have separate settings. As a result, the walls next to the cutting mesh will stop and loop around. There will also be skin generated between the original mesh and the part that was cut off. I believe you are seeing these effects and thinking they are bugs.

> Another user commented in the FB Group:
> Yup, having that issue with shells and modifier blocks, where Cura decides the shells should somehow end and not continue over the boundary despite the [...]

Use an infill mesh then. Giving the infill mesh a Wall Line Count of 4 will give this effect.

Your second-to-last screenshot shows what seems to be a cube above the model, and a diagonal edge beneath it. I can't tell if the modifier mesh extends that way downwards (if it's not a cube). If it is a cube, this issue could just be that you have Make Overhang Printable enabled. In that case, this is actually a bug, but a duplicate of https://github.com/Ultimaker/Cura/issues/9350 .

Is your modifier mesh a cube, and is Make Overhang Printable enabled?

closed 2021-04-29 by no-response[bot]