Ultimaker/Cura #9495: Cura Material Profile Duplicate/Remove Issue

WeslyJacobs, 2021-04-02:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

1. Open Preferences > Configure Cura...
2. Browse to Materials section
3. Duplicate Material Profile (e.g. Generic CPE) 

### Actual results

The Duplicated Material Profile can't be removed ("Remove" button is Greyed Out). When a subsequent Material Profile is Duplicated, the previous Material Profile can be removed again, but the newest Material Profile can't be removed ("Remove" button is Greyed Out).

### Expected results

Duplicated Material Profiles can be removed immediately after creating them.

[image]

fvrmr, 2021-04-02:
Hi @WeslyJacobs thank you for your report.
If you activate an other material, then you can remove the duplicated material. 
The reason why is because the material is activated and then you can't delete it. 

Let me know if you activate an other material that you can delete the duplicate.

3DPKid, 2021-04-02:
> If you activate an other material, then you can remove the duplicated material.
> The reason why is because the material is activated and then you can't delete it.

=> Thanks for the clarification, this wasn't clear to me, but sounds logical. Meanwhile I tested this out and this is effectively the case.

fvrmr, 2021-04-02:
Nice, then I will close this issue. Since this is solved!

closed 2021-04-02 by fvrmr