Ultimaker/Cura #9499: unable to start up cura 4.8 on windows 10

Callsignprofessor, 2021-04-04:
### Application Version

4.8.0

### Platform

windows 10

### Printer

creality ender 3 pro

### Reproduction steps

I did a clean windows 10 install on a new SSD and when reinstalling cura it shuts down when selecting a printer. tried running it as administrator and a bunch of other stuff but eventually installed 4.6.2. But now I can't import my cura profile (originally made on 4.6.2) 

### Actual results

It shuts down

### Expected results

cura should load the offline printers and when you select your printer cura should be operationall

jellespijker, 2021-04-04:
Please provide log files.
Goto help -> configuration  folder and the log files should be there

Callsignprofessor, 2021-04-07:
@jellespijker sorry for my delayed reply, ive checked the config file but it gave the reason "the version is outdated, its 16 and should be 14"

[cura.log](https://github.com/Ultimaker/Cura/files/6272047/cura.log)

jellespijker, 2021-04-07:
Those should be upgraded by the upgrader plugins. If you look in the marketplace are the upgrader plugins enabled?

Callsignprofessor, 2021-04-10:
I've checked with the marketplace upgrader plugins. Haven't seen anything that has that exact name but everything in 'marketplace>installed' is checked appart from 4.6 to 4.7 version upgrader.

Callsignprofessor, 2021-04-10:
So some miracle happend and 4.8 works on my pc! (checked everything but couldn't find the original issue) Profile is loading as well. Thanks @jellespijker for your help :)

closed 2021-04-10 by Callsignprofessor