Ultimaker/Cura #9500: unable to import cura profile

Callsignprofessor, 2021-04-04:
### Application Version

4.6.2

### Platform

windows 10

### Printer

creality ender 3 pro

### Reproduction steps

Install cura and imported cura profile and it gave a message of successfully imported but it doesn't show on the list. I can make a new profile and export/import it without issues.

### Actual results

It gives a popup "profile successfully imported" but doesnt show in the list

### Expected results

my profile in the list

MisterPrintf, 2021-04-04:
Not a bug; imported profiles will only show under the printer they have been designed for in most circumstances.

Callsignprofessor, 2021-04-05:
Yes I know, but it still doesn't import.

jellespijker, 2021-04-05:
Can you provide log files and profile files that want to import

Callsignprofessor, 2021-04-07:
Yes! sorry for my delayed reply.

[cura.log](https://github.com/Ultimaker/Cura/files/6272054/cura.log)

closed 2021-04-10 by Callsignprofessor

Ghostkeeper, 2021-04-15:
The log file indicates that you're trying to import a profile created by Cura 4.8 into Cura 4.6.2. 4.6.2 is unable to read the more modern format of profiles (because that simply didn't exist yet when 4.6.2 was released).