Ultimaker/Cura #9502: Cura does not effectively utilize available resources

MisterPrintf, 2021-04-04:
**Application version**
4.8.0

**Platform**
Windows 10, 64-bit
Intel i5-10600K overclocked to 5.1Ghz
Nvidia RTX 2060, overclocked to +100Mhz cores, +700Mhz memory
16GB DDR4-3200Mhz RAM overclocked to 3400Mhz

**Printer**
Ender 3

**Reproduction steps**
Slice a detailed model on a decently powerful machine, having CoreTemp, Afterburner and/or Task Manager open to monitor CPU, GPU and memory activity. Nowhere can a bottleneck be found; not even a single core is pushed beyond 30% utilization, databuses are nowhere near their cap, RAM is once again not anywhere near full, et cetera.

**Expected results**
Cura should utilize the available system resources to the fullest. While multithreading is of course not possible due to the linear nature of slicing, at least one CPU core should be running at above 90% utilization, or a bottleneck should be visible elsewhere (such as a databus).

jellespijker, 2021-04-05:
Could you provide log files and project files.
Slicing a fairly complex model on my Linux PC i7 seems to utilize all cores when possible. See screenshot the arrows indicate the core usage during the slicing.
[image]

MisterPrintf, 2021-04-13:
I ended up succesfully resolving a good portion of this by tweaking my motherboard's settings; it turns out the shitty OC software took over my manual settings and throttled some instruction sets to 2.5Ghz. That being said, while it is better it's still not perfect, as demonstrated by a snippet from Task Manager:

[image]

I'm unsure at this point whether the inconsistent spikes are due to Cura or my machine at this point; if you cannot reproduce them feel free to close the issue.

I'm unable to upload the project I'm using for testing, as it goes over the 10mb limit even when doubly compressed into .7z and then .zip.

Here is a shot of the model; the original creator seems to have taken it down from Thingiverse for some reason.

[image]

And lastly - I'm unsure whereabouts I would find the log file. Mind giving the location?

Ghostkeeper, 2021-04-15:
Where to find the logs: https://github.com/Ultimaker/Cura#logging-issues
On Windows with Cura 4.8, that would be `%APPDATA%\cura\4.8\cura.log`

There is a known issue that you could be seeing with tree support, if you have that enabled: https://github.com/Ultimaker/Cura/issues/6066

If it's too big for Github, you could consider uploading it elsewhere and posting a link (like WeTransfer or Mega). If it's tree support though, there are sufficient reproduction steps in the other ticket already.

closed 2021-05-17 by no-response[bot]