Ultimaker/Cura #9504: Sync all settings between some PC's

userosos, 2021-04-05:
### Is your feature request related to a problem?

Problem with sync between some PC's.

### Describe the solution you'd like

I want sync changes in an material, an profile, an addons between some PC's. For example, i changed print speed in the Profile on PC1 after it i goto PC2 and click an button (Sync) or auto and i can sow the new settings in the profile. If i changed an settings in PC2 i want see the settings in the PC1. 

### Describe alternatives you've considered

Cura backups not usability for it.

### Affected users and/or printers

I think it will be very comfortable for users who use more than 1 PC's. It make portable workplace for 3d printing.

fieldOfView, 2021-04-05:
> Cura backups not usability for it.

Please elaborate. Why is Cura Backups not a solution for you? What is wrong with "usability"?

userosos, 2021-04-05:
> 
> 
> > Cura backups not usability for it.
> 
> Please elaborate. Why is Cura Backups not a solution for you? What is wrong with "usability"?

It requesting restart, not stable (i can see crashing after restore backup sometimes), need a lot of time (i need make backup on PC1 after it i need start Cura on PC2 restore backup, restart Cura if i see crash i need fix it). I want just sync all settings but not replaced them. It is more simple and will be good i think.
For example add the function for local network. Sync all settings between the PC from local network. Or an options "Sync all settings from PC2 uses local network".
I used an other slicer with sync functions and i can see that it great.

rburema, 2021-04-07:
> (i can see crashing after restore backup sometimes)

We'd really like information about these crashes can you share your logs?

Otherwise, we've discussed this & our other priorities, and it seems very unlikely that we'll have any time to spare to work on the actual feature request part. We'll have to defer this. Sorry!

(Not closing this yet because of the crashes, which might lead to the discovery of a bug, and will potentially improve the usability of the backups for scenarios such as these.)

matthiasamberg, 2022-06-30:
why do you close this request. This request wasn't about software crashes, but about having a system that actually makes ultimaker printers usable from multiple computers. Manually syncing cura backups between different machines is a ridiculous workaround, why does this even need explanation?

fieldOfView, 2022-06-30:
The issue was closed because further information that was requested (logs) was never provided.

rburema, 2022-07-01:
@matthiasamberg Please read the replies before commenting :-)

You wrote today:
> why do you close this request. This request wasn't about software crashes, ...

I wrote on the 7th of April (see above):
> ... Otherwise, we've discussed this & our other priorities, and it seems very unlikely that we'll have any time to spare to work on the actual feature request part. We'll have to defer this. Sorry! ...

As always, pull requests welcome, of course. Or making use of and/or writing plugins, etc.

rburema, 2022-07-01:
Also note that 'deferred' doesn't mean closed of course! It was auto-closed by the bot, but @Ghostkeeper re-opened it, as it's proper, as at some point in the unspecified future we may indeed have time/prio to work on the feature request part of this issue.

closed 2023-07-15 by github-actions[bot]