Ultimaker/Cura #9505: Missing initial bed and print temp values on the filament profile form

Sigval, 2021-04-05:
### Is your feature request related to a problem?

having the ability to create our own print profiles are great :) 
having said that, it is awesome that we now also may create our own filament profiles on top of this. 
This is excellent when you want to use the same profile but different filament. 

However it is really frustrating that there is no value input field on the filament profiles for initial bed and initial print temp.
This means that I have to define these values directly within the print profile itself instead, meaning that I have to create one profile for each filament type rather than just the one and just selecting another filament profile. 

### Describe the solution you'd like

please add initial bed and print temp input fields on the filament profile form which will dynamically update the current print profile's corresponding field depending on the current selected filament profile. 

### Describe alternatives you've considered

Or alternatively at least just lower the print bed temp when filament profile type is PLA. currently both initial bed and bed is set to 60 for PLA by default, but keep in mind that PLA actually does not require any heated bed, for PLA the benefit of a heated bed is only at the beginning of the print for the initial layer to stick to the bed so you do not have to use glue. 

For example 60 for initial and 50 for print is already more than enough, if not just turning it of after initial layer. 
When it is left on 60 for the whole print I often experience warping, but if I lower the temp all over including the initial layer then the print won't stick well. so at least for me, 60 initial then down to 50 during print seems like better default values.

### Affected users and/or printers

I believe everyone will benefit from this

nallath, 2021-04-06:
Did you already had a look at the material settings plugin? You can find it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MaterialSettingsPlugin

Sigval, 2021-04-08:
Thanks I tried this, it helped a lot, exactly what I was looking for.
Though I didn't see any changes at first, I had to go to extensions and
then select material settings,
and check those additional settings I would like to visualize on the
material pane.

closed 2021-04-08 by nallath