Ultimaker/Cura #9506: Slicing stops half way through with Arachine but works fine with 4.8

Anathema-Device, 2021-04-05:
### Application Version

arachine_engine_beta

### Platform

Windows 10 - Surface Pro 4

### Printer

Geeetech A10M

### Reproduction steps

I have placed 3 parts on the table, selected supports from the bed, etc and clicked go.

### Actual results

Slicing stops 1/2 way through after a single processor hits 100% utilization by arachine.  There are no other programs other than the task manager running.

### Expected results

I expected slicing to complete

[gets_half_way_and_stops.zip](https://github.com/Ultimaker/Cura/files/6258472/gets_half_way_and_stops.zip)
[fails_half_way.cura.log](https://github.com/Ultimaker/Cura/files/6258473/fails_half_way.cura.log)
It seems to slice just fine with the same settings in Cura 4.8

rburema, 2021-04-06:
Hi @Anathema-Device 

Did you save the project file or export it? If I open it here, it doesn't ask me to open on a specific printer, and there are no settings in the file.

If you export the file, it will not be a Cura-project, but just a 'plain' 3mf. In order to reproduce this, I think we'll need a Cura-project file (either that or it was fixed after the last beta, but I doubt that).

Anathema-Device, 2021-04-06:
Hi @rburema,
I saved it as a "project" and it seems to have saved some settings as there was a notice about the changes when I opened the project with Cura 4.8.  How should I save it, or get the settings for you?
The version shown is simply "Arachne_engine_beta", it does not show a build date or anything to identify it further.
What file format under Export should I select?

rburema, 2021-04-07:
'Save project ...' should be the right one.

Anathema-Device, 2021-04-09:
Hi @rburema,
I am sorry to say that I no longer have the original project so I can't seem to reproduce the hang any longer.  It would stop about 1/2 way through the slicing progress bar then the blue bar would go away with no errors or anything like it just gave up.  So I will close this.  Sorry for all the trouble, I should have saved the project in a non-temp directory.

closed 2021-04-09 by Anathema-Device