Ultimaker/Cura #9510: Smarter bridging

Matszwe02, 2021-04-06:
### Is your feature request related to a problem?

When there's a hole and at some point it contracts to a smaller diameter, cura will brigde in one direction, causing half of the bridges (that ones that are not supported at both sides, so ecery line that ends at the smaller hole) to fall.

### Describe the solution you'd like

The solution will be to bridge lines that can be bridged, than rotate 45 degree, bridge what can be bridged, and so on, creating some shape like octagon, then printing the perimeter supported by these bridges.

### Describe alternatives you've considered

Alternatives are sacrifitial layers or supports.

### Affected users and/or printers

every FDM printer is affected.

fvrmr, 2021-04-13:
Hi @Matszwe02 thank you for your feature request.
Could you provide me some more information like a visual or a model were this happens. 
Thank you in advance.

Matszwe02, 2021-04-13:
[image]
I hope this will be easy to read and understand :)

Ghostkeeper, 2021-04-15:
I'd consider this to be a duplicate of https://github.com/Ultimaker/Cura/issues/7162 .

The idea is nice, but it doesn't work if the hole in the layer is not convex. We'd need some sort of extension that makes it generate a result for all polygons.

closed 2021-04-15 by Ghostkeeper

Matszwe02, 2021-04-15:
> I'd consider this to be a duplicate of #7162 .
> 
> The idea is nice when applied by a human, but it doesn't work if the hole in the layer is not convex. We'd need some sort of extension that makes it generate a result for all polygons. It has a lot of potential for better bridging, but needs more development.

I think this is simpler solution than this previous issue, because the slicer tries to bridge at one direction, then at the other direction, and makes every bridge possible for this direction. The previous method is indeed confusing for computers to understand (because it uses the shapes of these holes), but I think that even in the beta version it will be great to bridge one direction, then perpendicularly (that's better for computers, simply draw every bridge line, subtract these ended in mid air, then bridge the rest with the same principle), and the rest of bridging will be filled with normal lines or skipped for this layer.

Ghostkeeper, 2021-05-03:
It would be good to add that to the existing discussion, then.