Ultimaker/Cura #9511: Cura will not finish slicing stl

dtwolfen, 2021-04-07:
### Application Version

arachne

### Platform

mac 10.15.7

### Printer

artist d

### Reproduction steps

opened stl, added support blockers, changed some things in support and tried to slice, it slices for about half way with the blue line then the blue line disappears. I tried moving and rotating the stl with no improvements

### Actual results

The slicing stops after the blue line reaches half way and goes grey

### Expected results

slice and save a gcode

[cura.log](https://github.com/Ultimaker/Cura/files/6268462/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/6268464/cura.log)

nallath, 2021-04-07:
Could you also add the project file. We know that there are some models that cause engine crashes, but we do need the model in order to reproduce it.

dtwolfen, 2021-04-07:
[open button ideamaker.3mf.zip](https://github.com/Ultimaker/Cura/files/6270787/open.button.ideamaker.3mf.zip)

I think this is what you are asking.

dtwolfen, 2021-04-10:
I found that it slice just fine if I assign the extruder 2 to the stl and extruder 1 to support.

Ghostkeeper, 2021-04-15:
That's it, indeed. Thanks for the project file. I can reproduce the issue then. When a CuraEngine compiled in debug mode, it produces these log entries:

```
2021-04-15 19:54:18,118 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] There must have been at least one direction in which the bead count is increasing enough for the transition to happen!
2021-04-15 19:54:18,118 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [WARNING] The edge we're about to connect is already connected!
2021-04-15 19:54:18,319 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] *** stack smashing detected ***: terminated
2021-04-15 19:54:18,436 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [925]: Backend quit with return code -6. Resetting process and socket.
```
Stack smashing... Uh oh!

closed 2024-12-08 by github-actions[bot]