Ultimaker/Cura #9512: Cura crashes at welcome. Fatal Python Error Segmentation

DONNMARQ, 2021-04-07:
### Application Version

4.7

### Platform

WINDOWS 7 HOME PREMIUM

### Printer

DIDNT GET THERE

### Reproduction steps

I TRIED TO INSTALL 4.8 FIRST KEPT GETTING SIMILAR FATAL PYTHON ERRORS. SOME USB ERRORS SO I REMOVED MY USB MICRO SD READER AND BACK TO THIS OTHER ERROR LOG

### Actual results

SAME REPEATING ERRORS EVEN WITH VERSIONS 4.7 OR 4.8

### Expected results

CURA SHOULD LOAD UP

2021-04-06 22:45:34,353 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ConsoleLogger
2021-04-06 22:45:34,383 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraDrive
2021-04-06 22:45:34,423 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [92]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.7\CuraEngine.exe
2021-04-06 22:45:34,423 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraEngineBackend
2021-04-06 22:45:34,443 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileReader
2021-04-06 22:45:34,443 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2021-04-06 22:45:34,713 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura 4.7\CuraEngine.exe
2021-04-06 22:45:34,723 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.7\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2021-04-06 22:45:34,743 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileWriter
2021-04-06 22:45:34,769 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJReader
2021-04-06 22:45:34,797 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJWriter
2021-04-06 22:45:34,822 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLReader
2021-04-06 22:45:34,925 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLWriter
2021-04-06 22:45:34,945 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2021-04-06 22:45:34,956 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-06 22:45:34,962 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.7.0
2021-04-06 22:45:34,968 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2021-04-06 22:45:34,973 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-06 22:45:34,978 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2021-04-06 22:45:34,984 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2021-04-06 22:45:34,989 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2021-04-06 22:45:34,994 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2021-04-06 22:45:34,997 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-06 22:45:34,997 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2021-04-06 22:45:35,007 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2021-04-06 22:45:35,007 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2021-04-06 22:45:35,017 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2021-04-06 22:45:35,017 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-06 22:45:35,017 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2021-04-06 22:45:35,027 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2021-04-06 22:45:35,027 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2021-04-06 22:45:35,037 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2021-04-06 22:45:35,037 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2021-04-06 22:45:35,057 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2021-04-06 22:45:35,067 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2021-04-06 22:45:35,087 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2021-04-06 22:45:35,097 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2021-04-06 22:45:35,131 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2021-04-06 22:45:35,147 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2021-04-06 22:45:35,172 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2021-04-06 22:45:35,186 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2021-04-06 22:45:35,199 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2021-04-06 22:45:35,216 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2021-04-06 22:45:35,232 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
2021-04-06 22:45:35,240 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ModelChecker
2021-04-06 22:45:35,259 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MonitorStage
[... 197 lines cut ...]
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002a84 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00000af8 (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 844 in run
  File "X:\4.7-exe\build\inst\bin\cura_app.py", line 234 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

nallath, 2021-04-07:
Did you also try the 4.9 beta?

DONNMARQ, 2021-04-08:
yes still crashes. heres the log:
trimesh\util.py:2064: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.9 it will stop working
networkx\readwrite\graphml.py:346: DeprecationWarning: `np.int` is a deprecated alias for the builtin `int`. To silence this warning, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
networkx\readwrite\gexf.py:220: DeprecationWarning: `np.int` is a deprecated alias for the builtin `int`. To silence this warning, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
trimesh\voxel\runlength.py:205: DeprecationWarning: `np.bool` is a deprecated alias for the builtin `bool`. To silence this warning, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
2021-04-07 21:36:24,943 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin ConsoleLogger
2021-04-07 21:36:24,973 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraDrive
2021-04-07 21:36:25,023 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [96]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.9.0\CuraEngine.exe
2021-04-07 21:36:25,023 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraEngineBackend
2021-04-07 21:36:25,043 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraProfileReader
2021-04-07 21:36:25,043 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2021-04-07 21:36:25,053 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura 4.9.0\CuraEngine.exe
2021-04-07 21:36:25,063 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.9.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2021-04-07 21:36:25,073 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin CuraProfileWriter
2021-04-07 21:36:25,093 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin OBJReader
2021-04-07 21:36:25,093 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-07 21:36:25,093 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.9.0-BETA
2021-04-07 21:36:25,103 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2021-04-07 21:36:25,103 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-07 21:36:25,103 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2021-04-07 21:36:25,103 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2021-04-07 21:36:25,113 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2021-04-07 21:36:25,113 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2021-04-07 21:36:25,113 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-07 21:36:25,123 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2021-04-07 21:36:25,123 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2021-04-07 21:36:25,123 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2021-04-07 21:36:25,123 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2021-04-07 21:36:25,133 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2021-04-07 21:36:25,133 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2021-04-07 21:36:25,133 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2021-04-07 21:36:25,143 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin OBJWriter
2021-04-07 21:36:25,143 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
[... cut ...]
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\SolidView\SolidView.py", line 212 in beginRendering
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\Bindings\MainWindow.py", line 238 in _render
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

Ghostkeeper, 2021-04-15:
This looks like a duplicate of https://github.com/Ultimaker/Cura/issues/9456 . They also report that it's also crashing with the older version, and it's in the same point during start-up according to the log. Since you grabbed the log from the terminal there's a bit more information this time, which helps!
There is also a possible workaround in that thread; setting your GPU to graphics mode rather than compute mode. Not a fix, but a workaround.

Seeing as how we're suddenly getting a bunch of this crash reported in various sources, I think that a recent graphics driver version might be the culprit here. Perhaps a bug in the AMD drivers?

Maybe this same bug is causing your caps lock to be stuck as well. Should look into that.

closed 2021-04-15 by Ghostkeeper