Ultimaker/Cura #9514: Eraser object slicer error.

MartinSaldert, 2021-04-07:
### Application Version

4.8.0

### Platform

Win10, RTX 2080

### Printer

Zaribo (Prusa clone)

### Reproduction steps

Imported a cube STL.
Added a Eraser object
Set it to:
Modify settings for overlay
Cutting mesh
Top layers: 0

Slice...

### Actual results

I get no top layers were the eraser object is intersecting. = Perfect
But i do get a floor and walls on the eraser object so the cube is not open all the way down.

### Expected results

I would expect it to print as usual but skip the top layers were the eraser object intersects.
Not add a floor and walls were the eraser object intersects.
PrusaSlizer got it correct.

[PI3MK3M_cube.zip](https://github.com/Ultimaker/Cura/files/6269843/PI3MK3M_cube.zip)

nallath, 2021-04-07:
I don't know what prusaslicer does, but the behavior that you describe is as per design.

A cutting mesh, as the name implies, cuts up the model. These resulting models are essentially treated as separate models. As such, all of them will have top & bottoms.

I can't really come up with a reason why it should work as you describe. Could you elaborate a bit more as to what the usecase of that is?

MartinSaldert, 2021-04-07:
Hi.

I will try to explain 😊
I needed to print some parts without top layers to be able to fill the parts with resin to use on my boat.

The part had holes for bolts in it and I needed the bottom of the bolt holes to be printed but not the top layers.

A simple explanation is the stair image attached. I don’t need the top layer on the top step (red) but I do need the top layers on the lower step (Green)so I can pour a liquid in it and fill it to the top.

 

The “fake” image is what I was looking for. Bolt holes have a floor and no top layers. This image is from the preview when I removed the 4 last layers.

If I turn of top layers the bolt holes gets no floor since it’s a “top layer” and the resin will flood the bolt hole.

 

The Pruza image is what PrusaSlizer gave me. The top 2mm have a custom setting with no top layers. So I get a open model but still have a floor in the bolt holes (Red layer).

 

So I guess I was expecting the eraser object to only act as a “within this area” were I could change settings. Like it does when I use it to remove support.

 

/Martin Saldert

nallath, 2021-04-07:
> Hi. I will try to explain blush I needed to print some parts without top layers to be able to fill the parts with resin to use on my boat. The part had holes for bolts in it and I needed the bottom of the bolt holes to be printed but not the top layers. A simple explanation is the stair image attached. I don’t need the top layer on the top step (red) but I do need the top layers on the lower step (Green)so I can pour a liquid in it and fill it to the top. The “fake” image is what I was looking for. Bolt holes have a floor and no top layers. This image is from the preview when I removed the 4 last layers. If I turn of top layers the bolt holes gets no floor since it’s a “top layer” and the resin will flood the bolt hole. The Pruza image is what PrusaSlizer gave me. The top 2mm have a custom setting with no top layers. So I get a open model but still have a floor in the bolt holes (Red layer). So I guess I was expecting the eraser object to only act as a “within this area” were I could change settings. Like it does when I use it to remove support.

Attaching items via email doesn't work. You have to do it via the web interface.

MartinSaldert, 2021-04-07:
[image]
[image]
[image]

Images attached!

MartinSaldert, 2021-04-08:
Do you still need more info?

Ghostkeeper, 2021-04-15:
No, I think I understand. Your project file makes it quite clear, really. I think it's a bit of a misunderstanding of how cutting meshes work.

I've modified your project file a bit for demonstration. Here's what is happening in the cross section of your project file:
[image]

The cutting mesh cuts the original mesh in two pieces. One is still being printed with the global / per-extruder settings, where you had the Top Layers set to 4. The other has one setting overridden: Top Layers is set to 0. So the small part at the top is now a rectangular mesh without any top layers. The infill is exposed at the top. The remainder now has a gulley cut out at the top, but that still prints with top layers.

In Cura, a cutting mesh is just that. It cuts the model into two models. Both of these models will get walls and skin as normal, so there will be a border of the Bottom Layers of the cut part and the Top Layers of the remainder.

To get the effect that I think you desire, I suggest to set the Top Layers in the normal settings list to 0, then use an infill mesh with 4 top layers in the rest. The result is this:
[image]

Here's a project file to show what I mean: [PI3MK3M_cube_infill_meshes.zip](https://github.com/Ultimaker/Cura/files/6320603/PI3MK3M_cube_infill_meshes.zip)

Ghostkeeper, 2021-04-15:
Since this is expected behaviour, we'll close this issue. But if you have suggestions as to how we could've made this behaviour more clear, I'm still curious to hear them!

closed 2021-04-15 by Ghostkeeper