Ultimaker/Cura #9515: Cura Start Up Times and Port Access Annoyance

Legsmaniac, 2021-04-07:
### Is your feature request related to a problem?

I really do like Cura but the biggest annoyance for me is the Start Up time, some 45 seconds to open up my selected .stl when other slicers like Prusa starts up instantly.  

Another annoyance for me is the fact that Cura seems to probe my ports when starting up and during running.  For example, I often have some Arduino's plugged into my PC USB ports on test and when Cura is started, the Arduinos reset and reboot.  Every time!  They also reset/reboot/flash/flicker intermittently while Cura is up and running. SO annoying!  Other slicers such as Prusa doesn't do this.

### Describe the solution you'd like

I would like to see Cura start up MUCH faster than it does now.

I would also like to see Cura STOP probing my ports when starting up and running.

### Describe alternatives you've considered

It was suggested I try Barbarian Plugin but this does nothing for me.  Doesn't make any difference at all.

I therefore resort to alternative slicers such as Prusa Slicer, Super Slicer, Maker etc. and only use Cura if I need to, such as wishing to use Tree Supports which is sadly lacking in Prusa.

### Affected users and/or printers

I imagine this affects everyone  although a) not everyone is bothered about speed as I am (Time is money!!) and b) not everyone would have critical things plugged into their USB Ports during Cura use.

If Start Up Time and Port Probing could be improved/eliminated, I would use Cura so much more than I currently do.
Alas, for now, it's Prusa,  for me at least.  Sorry.

nallath, 2021-04-07:
I don't understand why anyone would say that thte barbarian plugin would change _anything_ with regards to USB ports or startup time. For the next time, could you also create seperate tickets for each issue?

There are a few things that you can do though. First off; Disable the USB printing plugin. That is the plugin that tries the USB ports to see which one has printers connected.

Secondly, we are looking into speeding up the booting time, but it's going to need rather big changes. Up to that time you can use the startup optimiser plugin.

Legsmaniac, 2021-04-07:
> I don't understand why anyone would say that thte barbarian plugin would change _anything_

> Up to that time you can use the startup optimiser plugin.

My bad.  I meant the Startup Optimiser.  Apologies.  Doesn't do anything at all.

> For the next time, could you also create seperate tickets for each issue?

Sigh.  OK, will do, if I remember.  Apologies again.  Seems I can't win.  Some Gits say don't keep posing sep deets and now this.  

> Disable the USB printing plugin

I wish it was that simple.  I don't have any USB Printing plugin.  Never have done.

Thanks for the reply though.  I'll look fwd to a future version of Cura which starts as fast as other Slicers.

GregValiant, 2021-04-09:
Just for a quick check, go to the Marketplace and Installed and scroll down to USB printing and make sure the box isn't ticked.  It is installed by default and enabled by default.

[image]

Legsmaniac, 2021-04-09:
@GregValiant Finally!!!!  Thank you, thank you, thank you, thank you, thank you!!
Seriously - thank you!
That's fixed it!  No more probing ports!  My Arduinos on test can run without flashing, beeping, resetting, rebooting!
Brilliant!
You are the first person to have actually helped me!  Although I do recall someone else suggesting I check if I have something like that installed but they didn't explain where to look like you did.  I just looked under "Extensions" which of course doesn't show that.  After all the Forums, Groups, Facebooks etc. I have posted for help on this annoyance, it finally gets rectified!

And what's more, I took a proper look through that list and unchecked a few other things I felt I didn't need/was never gonna use and it seems I have got my Start Up time down to somewhere between 11-12 seconds!!  Still 10 seconds longer than Prusa but I'll take that!!  Brilliant!

Oh, did I say Thank you?!?  😁      👍

GregValiant, 2021-04-09:
Your very welcome.  Glad to have been of assistance.

Now quit fooling around and get back to work.

Legsmaniac, 2021-04-09:
@GregValiant Work? It's bed time. Only woke up for my pills. Back to bed. Sleep while the printer is printing. 😁
Cheers buddy. 👍

nallath, 2021-04-09:
> You are the first person to have actually helped me! Although I do recall someone else suggesting I check if I have something like that installed but they didn't explain where to look like you did. [...]

This solution has been posted 10+ times on github itself already. Probably double that amount on all the unofficial help pages as well.

Legsmaniac, 2021-04-09:
Well they didn't come up in any searches, here or unofficial but never mind.  Sorted now thanks to Greg.

closed 2021-04-09 by Legsmaniac