Ultimaker/Cura #9517: About relative extrusion option in Cura 4.7.1

CirnoGiovanna, 2021-04-07:
### Application Version

4.7.1

### Platform

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

For some reason, I want generate gcode with relative extrusion (M83) by Cura. However, I do not found the corresponding useful options in Cura, or in def.json of CuraEngine. After I set the Relative Extrusion option visible in Cura, this option is still invisible in the Print setting palette.

### Actual results

Relative extrusion is invisible in the Print setting palette.

### Expected results

Relative extrusion is optional.

[image]

GregValiant, 2021-04-07:
In Ultimaker.def.json "Overrides":
"relative_extrusion":
        {
            "value": false,
            "enabled": false
        }
I'm not an Ultimaker guy but I would assume that override was added for a reason.

CirnoGiovanna, 2021-04-07:
Thanks for your comments.
I tried to use CuraEngine to slice the model. By using `-s relative_extrusion=true`, CuraEngine can generate gcode with relative extrusion. However, this gcode cannot be shown in Cura or print correctly in Ultimaker S3.

Ghostkeeper, 2021-04-15:
Tracking down the commit and issue where it started say that it's because

> There might be a firmware bug in the UM3 which causes problems with retractions when relative extrusion mode is enabled.

I didn't find a report of that firmware bug though. But indeed, it seems that the Ultimaker printers' firmware doesn't support relative extrusion at the moment. Perhaps it's got to do with the adjustments that the firmware makes to the g-code, to compensate for the bed being tilted or the compensation for slip in the feeder.

Ghostkeeper, 2021-04-15:
Since this is expected behaviour, I'll close this issue. Sorry! If the firmware doesn't support it, there's little that Cura can do.

closed 2021-04-15 by Ghostkeeper