Ultimaker/Cura #9520: Adding comments to settings profiles

printingotb, 2021-04-07:
### Is your feature request related to a problem?

I would like to be able to add a comments section to my custom settings profiles.

### Describe the solution you'd like

This would make it easier to remember what settings I want to investigate further and to maintain a change log. 
Or "this setting helped with that issue but I need to remember that it also affects this..."
If the comments also exports/imports with the profile, one could add instructions for others when sharing profiles.

### Describe alternatives you've considered

Keep a document on my computer, save new profiles every time I make a change. Include a readme document when sharing profiles.

### Affected users and/or printers

Anyone

[image]

fvrmr, 2021-04-13:
Hi @printingotb thank you for your feature request.
I will discuss this with the team. Keep you posted!

Glodigit, 2021-04-17:
A possible workaround for now is using the Material Settings plugin. I've put a large majority of my changed settings in the material now as most of the settings are material related in one way or another. As well as being able to compare settings faster since there's no lag like when you change profiles, you could then put the comments in the material comment sections.

printingotb, 2021-04-18:
That's clever but unfortunately I haven't found materials useful for me even with the plugin. I change a lot of settings to optimize for every print all the time including temp even for the same material. I instead use around 50 settings profiles, a few base profiles and one for every production part as they are all unique and the settings optimized. If I want to test a different value for a setting in a certain profile, I always save a new version so I can go back if it has a negative effect (I sometimes have more than one extra version). It would be simpler to write the change down in the profile instead.

fvrmr, 2021-06-15:
This issue is related to https://github.com/Ultimaker/Cura/issues/7468 and that issue has been deferred because we won't expect to work on this anytime soon. 

I will close this issue as a duplicate. Hope you understand.

closed 2021-06-15 by fvrmr