Ultimaker/Cura #9521: Colorblind Ass

mbourque, 2021-04-07:
### Application Version

4.9.0

### Platform

OSX

### Printer

Creality CR-6 SE

### Reproduction steps

1. Preferences
2. General

### Actual results

You should not refer to color blind people as "Colorblind Ass"

Also no matter if I change theme, nothing happens and setting does not stick.

### Expected results

Change terms to not refer to people as "Ass"
Make these themes settings work

[image]

mbourque, 2021-04-07:
I understand that for setting to take place, you must restart. If so, should notify user of this fact.

mbourque, 2021-04-07:
Ah - I totally missed that. Thanks

rburema, 2021-04-09:
Devs: See internal ticket CURA-8161

nallath, 2021-04-09:
Fixed by https://github.com/Ultimaker/Cura/pull/9545

closed 2021-04-09 by nallath