Ultimaker/Cura #9522: Setting language does not appear work

mbourque, 2021-04-07:
### Application Version

4.9.0

### Platform

OSX

### Printer

Creality CR-6 SE

### Reproduction steps

1. Preferences
2. General
3. Interface > Language

### Actual results

I change to English, but it does not seem to set and is blank when revisiting the Preferences

### Expected results

It should set and be remembered 

[image]

rburema, 2021-04-09:
Thanks for reporting this. We're aware of this ... it's part of a last minute fix to get the drop-down menu's to work at all on Mac  for the beta.

The language changing itself _should_ work however (that is, if you select say, Spanish in that box, after a restart, Cura should be in Spanish), it's just that the box now doesn't seem to listen to the currently active language. Which should definitely be fixed among the other things that are now wrong with the Preference Screen.

We should already have an internal ticket for this, I'll link it back here when I find it.

rburema, 2021-04-09:
Devs: See internal ticket CURA-8161

jonatanpolak, 2021-04-13:
When you click on select "language" something like this happens.
[image]
[image]
[image]

Ghostkeeper, 2021-04-15:
This has been fixed for the 4.9 stable release by https://github.com/Ultimaker/Cura/pull/9545:
[image]

closed 2021-04-15 by Ghostkeeper