Ultimaker/Cura #9524: requirements.txt has out of date versions

fieldOfView, 2021-04-07:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

any

### Reproduction steps

Run `pip install -r requirements.txt` with Python 3.8 (as per Cura 4.9)

### Actual results

Pip will fail to install the proper version of PyQt, because an older version is specified in requirements.txt

### Expected results

`PyQt5==5.15` is installed

It may we worth considering upgrading these dependencies:

```
numpy==1.19.5
scipy==1.5.4
shapely[vectorized]==1.7.1
```

Installing numpy, scipy and shapely on Windows is a bit more involved than just using pip and having things compile as needed. Therefore, many people use precompiled wheels instead. A popular source of these wheels is https://www.lfd.uci.edu/~gohlke/pythonlibs/. That site does not provide the requested version of the mentioned dependecies for Python 3.8.

Ghostkeeper, 2021-04-08:
Yes, this has been outdated for a while. It's a bit annoying to have to keep these in sync. I'd honestly rather have a single build system rather than two of which one is incomplete. This requirements.txt system assumes that everything is on Pip which is not the case for Cura's dependencies (but it does make things easier, if it's kept up to date, for a partial installation).

I've been meaning to update it once our keyring story has been completed, because apparently there is a security risk in the version of cryptography we were using. That is updated in the build now but our ticket hasn't been completed yet.

The CMake system is the ground truth here, so these are the packages that should be installed via Pip: https://github.com/Ultimaker/cura-build-environment/blob/master/projects/python_packages.cmake

Ghostkeeper, 2021-04-15:
I've updated requirements.txt here: https://github.com/Ultimaker/Cura/commit/8bc2c718a2ca0195f20f4f988ae58157c1575c54

Thanks for the help!

closed 2021-04-15 by Ghostkeeper