Ultimaker/Cura #9525: Cura hangs

mbourque, 2021-04-07:
### Application Version

4.9.0 Beta

### Platform

OSX

### Printer

Creality CR-6 SE

### Reproduction steps

Leave session open for an hour untouched

### Actual results

hangs

### Expected results

no hang

2021-04-07 12:45:38,625 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
2021-04-07 12:45:38,630 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SolidView
2021-04-07 12:45:38,645 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///Applications/Ultimaker Cura Beta.app/Contents/MacOS/resources/qml/PrintSetupSelector/Custom/QualitiesWithIntentMenu.qml:168:25: QML Binding: Not restoring previous value because restoreMode has not been set.
This behavior is deprecated.
You have to import QtQml 2.15 after any QtQuick imports and set
the restoreMode of the binding to fix this warning.
In Qt < 6.0 the default is Binding.RestoreBinding.
In Qt >= 6.0 the default is Binding.RestoreBindingOrValue.

2021-04-07 12:45:49,091 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-07 12:49:18,259 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-07 12:49:18,724 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-07 12:49:30,777 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-07 12:53:13,010 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura Beta.app/Contents/MacOS/resources/shaders/xray.shader]...
2021-04-07 12:56:38,069 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestTimeout [302]: Request [<UM.TaskManagement.HttpRequestManager.HttpRequestManager object at 0x600006fd9720>] timeout.
2021-04-07 12:56:38,073 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[e1517108][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError Operation canceled. The server returned: b''
2021-04-07 12:56:38,077 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [415]: request[e1517108][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] was aborted, do nothing
2021-04-07 12:56:38,078 - WARNING - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._onUserPackagesRequestFinished [74]: Requesting user packages failed, response code None while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')
2021-04-07 12:58:46,091 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-07 12:58:46,637 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-07 12:58:57,147 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-04-07 13:01:50,525 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestTimeout [302]: Request [<UM.TaskManagement.HttpRequestManager.HttpRequestManager object at 0x600006fd9720>] timeout.
2021-04-07 13:01:50,621 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[7ad4d311][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError Operation canceled. The server returned: b''
2021-04-07 13:01:50,624 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [415]: request[7ad4d311][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] was aborted, do nothing
2021-04-07 13:01:50,625 - WARNING - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._onUserPackagesRequestFinished [74]: Requesting user packages failed, response code None while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')
2021-04-07 13:05:49,065 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestTimeout [302]: Request [<UM.TaskManagement.HttpRequestManager.HttpRequestManager object at 0x600006fd9720>] timeout.
2021-04-07 13:05:49,073 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [376]: request[ae2ee22f][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] got an QNetworkReplyError Operation canceled. The server returned: b''
2021-04-07 13:05:49,076 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [415]: request[ae2ee22f][get][PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')][timeout=10][no-data] was aborted, do nothing
2021-04-07 13:05:49,077 - WARNING - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._onUserPackagesRequestFinished [74]: Requesting user packages failed, response code None while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v1/user/packages')
2021-04-07 13:08:08,090 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-04-07 13:08:08,579 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-04-07 13:08:20,748 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles

fieldOfView, 2021-04-07:
What version of MacOS do you use?

ChristopheBis, 2021-04-09:
I have the same situation on mac os big sur 11.2.3 macbook M1 and Cura **4.8**

Ghostkeeper, 2021-04-15:
The Qt platform at the moment doesn't support the Apple M1 yet. This seems to be the QTBUG tracking their progress: https://bugreports.qt.io/browse/QTBUG-85279

This is causing a wide variety of issues. Some people get it to run fine. For some people it crashes. For some it crashes in layer view. It's just altogether unreliable at the moment but there is very little we can do about it until Qt makes it work. And even then there's a chance that we'd need to go and compile the entire application for ARM to get it to work properly.

fvrmr, 2021-05-14:
Closing this issue since it is a duplicate of https://github.com/Ultimaker/Cura/issues/9710 so we keep everything in one issue.

closed 2021-05-14 by fvrmr