Ultimaker/Cura #9526: Material Import Failure

dcs3473, 2021-04-07:
### Application Version

4.9 Beta

### Platform

MAC OS

### Printer

Prusa Mini

### Reproduction steps

When manually importing new materials from the local hard drive I am able to import one successfully but then when you press the import button a second time it fails.

### Actual results

The button beeps and the dialog box never opens a second time.

### Expected results

The button should allow you to import more then one without having to restart.

nallath, 2021-04-08:
I can't reproduce this. Could you also share your log files (and if possible: the material files that cause this issue)

dcs3473, 2021-04-12:
Sorry for the delay.  Can you explain to me how to share my log files?

nallath, 2021-04-13:
https://github.com/Ultimaker/Cura#logging-issues

closed 2021-04-27 by no-response[bot]