Ultimaker/Cura #9527: Arachne. Not slicing (possibly defects in the file itself).

IGO-rek, 2021-04-07:
### Application Version

Arachne

### Platform

Windows 10

### Printer

DeltaBot

### Reproduction steps

1. Opened the file
2. sliced with custom settings

### Actual results

Nothing. it doesn't slice

### Expected results

There may be a problem with the file, but in theory it should have been sliced.

[image]
[Slising Arachne.zip](https://github.com/Ultimaker/Cura/files/6273607/Slising.Arachne.zip)

GregValiant, 2021-04-09:
I don't have a problem slicing the files in 4.8 or 4.9b.  After repairing the STL with MS 3D Builder it still sliced.  Using my printer and my profile on the STL it sliced.

konskarm, 2021-04-09:
Indeed it hangs in arachne while it slices without an issue in 4.9-beta. I'll raise this with the team.

ksihota, 2021-04-09:
Check your infill type and see if it is zigzag or line (with Connect Infill Lines checked)
I have several items that begin to slice but then never finish with these settings but work fine with other infill settings.
[image]
[image]

IGO-rek, 2021-04-10:
> «Соединить линии заполнения»

Thank you! It really works. I realized that at low fill values and with the "Connect Infill Lines" enabled, it is impossible to cut.

dtwolfen, 2021-04-14:
I was having this same problem wasn't until I saw this post that I realized how to get it to slice

Ghostkeeper, 2021-04-15:
We've put this on our backlog, targetting it for when Arachne gets released in a stable build. Developers, see issue CURA-8174.

ksihota, 2021-04-26:
I have found another time when arachne_engine_beta fails to complete a slice (progress bar disappears and slice does not complete.)
When the line width and number of walls hits a specific point where a new line will not fill a space this seems to happen. I can decrease the line width or decrease the number of walls and it will work, but a certain combinations the slice will fail.
[image]
[image]
[image]

Ghostkeeper, 2021-05-03:
@ksihota I can't reproduce your problem with my own testing models. It seems to complete the print fine every time, when I twiddle the line width around an integer number of walls. Do you have a project file to share that reproduces the issue?

ksihota, 2021-05-03:
@Ghostkeeper  Sorry, I deleted that project. I just did a quick check and it works with my current setup so there must be at least one other setting that is influencing this behaviour. I will try and find the settings where it was happening. If I can reproduce it I will upload the file here.

ksihota, 2021-05-03:
I went through all the wall settings and can't reproduce it. Maybe it was some other computer issue. Not sure why it was so specific to those 2 Cura settings though. If I discover the relationship I will post it here. Sorry for not investigating more before posting it.

closed 2024-12-08 by github-actions[bot]