Ultimaker/Cura #9528: [4.8.0] How does "make overhang printable" work?

akashgarg101, 2021-04-07:
Hello! I am looking to understand how the "make overhang printable" feature works in Cura. Any suggestions on how I can get to the source code that implements that feature? I understand that Cura is open source and the source code is available on Github, but I am having trouble finding the appropriate location for this part of the code. I am a PhD student and would like to implement a similar algorithm for my research. Apologies if this has already been answered elsewhere, but I looked and couldn't find anything. I know this is not a bug or issue, but I wasn't really sure where else to post this request. Would really appreciate any help! 

**Application version**
Cura 4.8.0

**Platform**
MacOS

**Printer**
Custom 3D printer

jellespijker, 2021-04-07:
Look in the CuraEngine for the actual implementation.
If you search for bridging on the commits there you get a sense of where that code resides. Which is unfortunately not always in a clear and single function.
https://github.com/Ultimaker/CuraEngine/search?q=bridging&type=commits 

Maybe one of my colleagues @rburema or @Ghostkeeper can steer you towards a more precise location.

Out of curiosity but what is your specific research topic?

BagelOrb, 2021-04-07:
The algorithm is super simple:

For every layer: 
- offset the outline inward
- union that to the layer below

akashgarg101, 2021-04-07:
Thanks so much @jellespijker and @BagelOrb. This is exactly what I was looking for! I am looking to apply a similar algorithm to bitmap-based 3D printing.

closed 2021-04-07 by akashgarg101

BagelOrb, 2021-04-08:
You can do a similar thing by applying an erosion to the bitmap of a layer and taking the max between that and the layer below as a type of union.

Do note that the method can still leave overhangs, though. Problems appear when a thin piece of geometry is eroded away fully.

I made an algorithm to deal with that in the code for conical support. Basically you can use boolean operations and offsets to determine the area which was removed, but which is too far from the previous layer. I don't have any algorithm to actually start reducing such thin geometry only from one end, though.

If you're interested, take a look at my papers: https://scholar.google.com/citations?user=wMabbjAAAAAJ&hl=nl&oi=ao

akashgarg101, 2021-04-08:
Thanks so much for the detailed response @BagelOrb . I was actually trying to do exactly what you described - an erosion based approach and ran into the problem that you describe - issues with handling thin geometries. I will take a look at your implementation of the conical overhang. 

I looked at your papers, it’s great and very interesting work! but I didn’t find anything where you describe the algorithm for overhang detection and fixing. Maybe I missed something?

BagelOrb, 2021-04-08:
Basically you can do something like:
```
insetted = layer_above.offset(-layer_height)
supported = insetted=offset(layer_height + rounding_error_term)
unsupported = layer_above.difference(supported)
layer_below = layer_below.union(insetted).union(unsupported)
```
Only thing is that this will leave those unsupported areas disconnected and unaltered for all layers below it.

No I haven't writen a paper on this specificly, but I thought you might be interested nonetheless ;)

I'm in the last year of my PhD, at TU Delft. How about you?

akashgarg101, 2021-04-08:
Ah ok, cool, got it! I am in my 3rd year of PhD at Carnegie Mellon University.