Ultimaker/Cura #9531: Unable to rename groups/"erasers"

KenLucke, 2021-04-08:
### Application Version

4.8.0

### Platform

Mac OS 10.15.7 (Catalina)

### Printer

N/A

### Reproduction steps

Group some objects
Attempt to rename the object group or support blocker ("eraser") via Mesh Tools > Rename

### Actual results

Rename Dialog appears, allows entry, but fails to rename group or support blocker.

### Expected results

Should allow all objects - models, support blockers ("erasers"), and groups - to be renamed, for clarity in the Objects List.

[claws.3mf.zip](https://github.com/Ultimaker/Cura/files/6275222/claws.3mf.zip)

nallath, 2021-04-08:
Please report issues with the meshtools plugin with the developer of that plugin: https://github.com/fieldOfView/Cura-MeshTools

KenLucke, 2021-04-08:
Thanks, it's hard to find the plugin Github pages.

nallath, 2021-04-08:
The marketplace shows the website for most plugins. That's where I got this from.

fieldOfView, 2021-04-08:
> That's where I got this from.

You had to look it up? For shame!

nallath, 2021-04-08:
Psh, you have so much plugins and I had the marketplace open :P

fieldOfView, 2021-04-11:
I looked into the issue of renaming groups using the MeshTools plugin, and it turns out that Cura renames the group right back to `Group #n` here:
https://github.com/Ultimaker/Cura/blob/master/cura/UI/ObjectsModel.py#L155
Shall I create a PR to make that code a bit more permissive of other names (as long as there is a name), or is there a reason to force the group names to `Group #n`?

Ghostkeeper, 2021-04-15:
Fixed link to where it gets renamed (permalink this time):

https://github.com/Ultimaker/Cura/blob/4bf440b86efef17a535299b93e82b3baefcaf16f/cura/UI/ObjectsModel.py#L171

Ghostkeeper, 2021-04-15:
This seems to originate from here: https://github.com/Ultimaker/Cura/commit/c839bc175998707ec1ddddb4464298ff2884d812

As far as I can tell, the purpose of this is just to make sure the groups have some recognisable/identifiable name, rather than an empty string or all having the same name when groups are created.

If the group names are unique by default, it's fine that they get given other names. So a pull request to fix this would be welcome! Otherwise I'll propose this to the team, but I must say that I don't expect our PO to give this a lot of priority as you might understand.

fieldOfView, 2021-04-16:
I'll create a PR that does something smart with the group names then.

closed 2021-04-28 by nallath