Ultimaker/Cura #9532: Cura 4.9.0 not recognizing USB

siquenique, 2021-04-08:
### Application Version

4.8.0 and 4.9.0

### Platform

Mac OS Big Sur - Macbook pro M1 

### Printer

Hellobeeprusa

### Reproduction steps

1. installed cura 4.8.0 and 4.9.0
2. added beeprusa printer
3. opened STL file
4. sliced

### Actual results

There is no option to print via USB although the printer looks being recognized by the system since i can use pronterface and control the printer.

### Expected results

Appear the option to print via usb instead of only save to disk

jellespijker, 2021-04-09:
Can you  provide us with a log file?

rburema, 2021-04-09:
(Just to be sure: I assume you'd have noticed the little arrow in the blue button that leads to other options.)

What does the monitor-tab show? In addition to the logs, if it's not giving a control panel there to the right of the screen, that's another indication that Cura hasn't found the printer.

siquenique, 2021-04-09:
Hello,

Here it is

Cheers,
Joao

siquenique, 2021-04-13:
Any feedback?

Ghostkeeper, 2021-04-15:
Hmm, well the only thing in that log being said about the USB printing plug-in is these two lines:

```
2021-04-08 09:39:21,240 - INFO - [MainThread] Toolbox.src.Toolbox.disable [454]: USBPrinting was set as 'deactive'.
2021-04-08 09:39:23,834 - INFO - [MainThread] Toolbox.src.Toolbox.enable [446]: USBPrinting was set as 'active'.
```

That's probably you trying to deactivate and reactivate the plug-in. I get where that approach is coming from :smile: But if it detects any COM ports I would expect it to say that it attempts connection with them at various baud rates.

One thing to look out for is to see if MacOS is just forbidding Cura from seeing the serial ports. That would be under System Preferences -> Security & Privacy -> Privacy tab -> Input Monitoring. If Pronterface is checked there, Cura would need to be too.

If you really want to get into it deep, this is a way you could emulate the way in which Cura detects serial ports:

```
> python3 -m pip install pyserial
> python3
>>> import serial.tools.list_ports
>>> serial.tools.list_ports.comports()
```

In my case (on Linux) that results in `[<serial.tools.list_ports_linux.SysFS object at 0x7f2fe25e8af0>]` because I have one SD card reader attached that talks via serial port.

closed 2021-05-17 by no-response[bot]