Ultimaker/Cura #9533: Can't perform retraction to stop stringing

Levy000, 2021-04-08:
### Application Version

4.8.0

### Platform

win 10

### Printer

ender3 pro

### Reproduction steps

1. I ticked “combing mode” （Also enable retraction is ticked）
2. I turned on “timelapse”

### Actual results

The printer doesn't perform retraction before time lapse in the later part of the printing process. This problem doesn't exist at the beginning of the printing process. So there is a lot of stringing in the head of models. 

### Expected results

A retraction always be performed before each time lapse.

[image]
[image]
[Minecraft_Guy.zip](https://github.com/Ultimaker/Cura/files/6277443/Minecraft_Guy.zip)

GregValiant, 2021-04-09:
It doesn't look like the Time Lapse plugin has any effect on retractions.  It doesn't appear to add them.  The retractions just happened to be wherever they were when the Time Lapse lines were added.  Combing being set to "All" may have had an effect.  Retract before outer wall may have had an effect as well.  Notice that the retractions stop occurring at layer 240 (241 in Cura).  That's where the last change in the part geometry occurs.
If you use "Travel | Retract at layer change" then you will get the retractions you want for the Time Lapse.

Ghostkeeper, 2021-04-15:
Indeed, the timelapse post-processing script only adds the commands for the timelapse.

I'll turn this into a feature request for the timelapse post-processing script to also have an optional retraction.

Ghostkeeper, 2021-04-15:
Actually, no. The timelapse post-processing script already has a retraction option. You've just set the retraction distance to 0. Give it a good bit of retraction (15mm?) and you'll be fine.

closed 2021-04-15 by Ghostkeeper

GregValiant, 2021-04-15:
Well @Ghostkeeper, you are right again.  The 4.9beta version of Cura has the retraction.  The 4.8 does though, but it's version can be easily over-written with the newer TimeLapse.py which I will include here in case @Levy000 doesn't care to install Cura 4.9.beta.
Over-write the existing TimeLapse.py file in "Program Files\Ultimaker Cura 4.8.x\plugins\PostProcessingPlugin\scripts" with the 4.9 version contained in the zip file.
[TimeLapse.zip](https://github.com/Ultimaker/Cura/files/6321215/TimeLapse.zip)