Ultimaker/Cura #9534: Custom Printer - Printhead Settings X min and Y min (XYZ - Da Vinci Nano)

mottainai-x, 2021-04-08:
### Application Version

4.8 and 4.9Beta

### Platform

MacOS Big Sur

### Printer

Custom FFF Printer

### Reproduction steps

Trying to setup a Custom Printer with the following settings: https://xyzofficialweb.blob.core.windows.net/xyz-v3/download/Cura_Profile_XYZ.PDF

Printhead Settings X min and Y min are specified as positive by XYZ but should be negative.
(I've input all other settings as advised)

### Actual results

Print head sits ~10mm from the build plate (Z axis) when printing.

### Expected results

Print head should sit appropriate distance from build plate.

There is very little documentation and support from XYZ and their XYZprint slicing/printing software is not as good as Cura.

GregValiant, 2021-04-08:
The Print Head settings describe the shape of the print head with 0 being the centerline of the nozzle.  When printing multiple models in "one-at-a-time" mode then Cura uses the numbers, else they are ignored.
The Home Offsets are set in the printer and determine where a print starts in relation to the Auto-Home position.  Your Home Offset X and Y are usually low negative numbers or 0, and your Home Offset Z should be 0.  If your prints are starting at 10mm above the plate then it sounds like the Home Offsets Z is at -10.

mottainai-x, 2021-04-08:
Ah, this makes sense, thanks @GregValiant 

It looks like I can change the Z offset in the XYZprint software. (Currently set at -4.21mm, Nozzle sitting on the build plate)
But only in increments of 0.05mm at a time by pressing the Z^^ button (ie: eventually moving it to -3.21mm, further away from the build plate).
[image]

Why then does it print fine from XYZprint but not from Cura?

Any help and advice is much appreciated. 

I did a quick Google search and it looks like Octoprint or Pronterface are good tools for changing my printers settings but after downloading Pronterface it comes up with "Printrun-Mac-18Nov2017" needs to be updated" error, and Octoprint looks very involved before you even get started...

GregValiant, 2021-04-09:
I don't know your printer or the XYZ slicer (although I have 17 slicers installed, that isn't one of them).  There are a couple of companies who insist that owners use their software to slice.  The printer firmware will be at least odd, and maybe proprietary, and only their slicer will put the code together in a manner that is acceptable to the printer.  I don't know that XYZ is one of them, I'm just sayin'.
Post a gcode file that the XYZ slicer came up with.  Printers only do what their told and maybe there will be a clue in the gcode as to what is going on.
Typically, after Auto-Homing, you move the print head to a position just in from the front left corner of the build surface and at a Z of 0 and then set the Home Offset at that position.  That's the 0,0,0 that the printer uses to set the location of a print.  On that dashboard things are different (the upper icons look like it's the settings for the Z offset for the Auto-Leveling?), but there may be a way to adjust the start gcode in Cura to achieve the proper starting Z.

mottainai-x, 2021-04-09:
How do I get the gcode file from the XYZprint slicer??...  

I found this post on SoliForum from David.Tucker [](http://www.soliforum.com/topic/17028/print-gcode-files-to-minimaker/)
MiniMaker is a similar XYZ printer to the Da Vinci Nano, and further down the page he posted edits he's made to the gcode header in Cura.. [](http://www.soliforum.com/post/141661/#p141661)

(XYZware is different to what I'm using (XYZprint) but I presume the gcode is still applicable. )

Is this a way of adjusting the Home Offset/Z Offset and how do I work out how much to set it by?
(Apologies in advance, I've been staring at a computer all day and my brain is fried!)

GregValiant, 2021-04-09:
This is from [HERE](http://www.soliforum.com/topic/6508/use-cura-with-da-vinci-printer/)...
"I was able to print successfully using Cura.  I compared the gcode generated by xyzware against Cura for the same STL file and noticed Cura use G0 command in gcode whereas xyzware doesn't.  So I replaced all G0 with G1 and voila, it printed beautifully **provided the usual xyzware header is included.**  There is no need to offset any z value.  The only downside is I had to manually replace G0 with G1 line by line and that took a long time even for a small file.  It would be nice if someone can write a little program that will automatically replace all G0 with G1."

I do believe this is one of those cases where the manufacturer definitely wants you to use their software.

Ghostkeeper, 2021-04-15:
There is a post-processing script "Search and Replace" that could automate that in Cura.

You'd make it search for `G0 ` (note the included space at the end) and replace that with `G1 ` (another space at the end).

The Cura developers can't provide support for specific printers. All I can really say is that the PDF you linked to is slightly outdated and the X Min and X Max should have a little minus symbol in front. Other than that it should still work the same as when the PDF was created.
The Cura issues are meant to report bugs and feature requests, but helping with configuring Cura to support certain printers is not something we can really do. I don't have all printers on the market in my home. I hope you can find some help among the community of XYZware!

closed 2021-04-15 by Ghostkeeper