Ultimaker/Cura #9540: Cura Not closing exit. force quit required

3D-Prints-4U, 2021-04-08:
### Application Version

4.9.0 beta

### Platform

Mac

### Printer

Ender # Pro

### Reproduction steps

I closed the program and it freezes. have to force quit every time I quit cura

### Actual results

I go on and use my Mac

### Expected results

For cura to shut down

rburema, 2021-04-09:
It would be really helpful to have logs here, and what specific Mac and OS (version) you're using. (Because this is not happening for our QA's on their Macs.)

The logs can be found in the Configuration Folder, (cura.log), which can be accessed by 'Help > Configuration Folder'.

3D-Prints-4U, 2021-04-09:
RON SHANNON
***@***.***
Stay Forever Young
Be Safe

3D-Prints-4U, 2021-04-16:
[image]

3D-Prints-4U, 2021-04-16:
cura.log

3D-Prints-4U, 2021-04-16:
Was this helpful?

Mr Ron

fvrmr, 2021-04-16:
Hi @3D-Prints-4U could you add your log in a comment on Github. You've replied with e-mail and then it won't attach your files.

3D-Prints-4U, 2021-04-16:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/6325926/cura.log.zip)

3D-Prints-4U, 2021-04-16:
that took forever

Ghostkeeper, 2021-04-20:
Thanks. Analysing that log file a number of weird things stand out to me:

```
Exception: Import error loading module CuraFreeCADPlugin
2021-04-16 09:32:44,458 - ERROR - [MainThread] UM.Logger.logException [116]: Traceback (most recent call last):
2021-04-16 09:32:44,463 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/UM/PluginRegistry.py", line 649, in _findPlugin
2021-04-16 09:32:44,467 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/4.9/install/lib/python3.8/imp.py", line 244, in load_module
2021-04-16 09:32:44,471 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/env/4.9/install/lib/python3.8/imp.py", line 216, in load_package
2021-04-16 09:32:44,476 - ERROR - [MainThread] UM.Logger.logException [116]:   File "<frozen importlib._bootstrap>", line 702, in _load
2021-04-16 09:32:44,491 - ERROR - [MainThread] UM.Logger.logException [116]:   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
2021-04-16 09:32:44,501 - ERROR - [MainThread] UM.Logger.logException [116]:   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
2021-04-16 09:32:44,508 - ERROR - [MainThread] UM.Logger.logException [116]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
2021-04-16 09:32:44,515 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/Luckys_Trevels/Library/Application Support/cura/4.9/plugins/CuraFreeCADPlugin/CuraFreeCADPlugin/__init__.py", line 12, in <module>
2021-04-16 09:32:44,519 - ERROR - [MainThread] UM.Logger.logException [116]:     from . import FreeCADReader
2021-04-16 09:32:44,525 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/Luckys_Trevels/Library/Application Support/cura/4.9/plugins/CuraFreeCADPlugin/CuraFreeCADPlugin/FreeCADReader.py", line 14, in <module>
2021-04-16 09:32:44,529 - ERROR - [MainThread] UM.Logger.logException [116]:     from .CadIntegrationUtils.CommonCLIReader import CommonCLIReader  # @UnresolvedImport
2021-04-16 09:32:44,534 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/Luckys_Trevels/Library/Application Support/cura/4.9/plugins/CuraFreeCADPlugin/CuraFreeCADPlugin/CadIntegrationUtils/CommonCLIReader.py", line 19, in <module>
2021-04-16 09:32:44,548 - ERROR - [MainThread] UM.Logger.logException [116]:     import objc
2021-04-16 09:32:44,556 - ERROR - [MainThread] UM.Logger.logException [116]: ModuleNotFoundError: No module named 'objc'
```
You seem to have a CuraFreeCADPlugin installed which crashes on start-up. Their website at thopiekar.eu is currently down, but this needs to be reported to the author, @thopiekar . Looking at Thopiekar's mirror of that plug-in on Github, it refers to a commit ID in CadIntegrationUtils which doesn't exist any more, and in the [current version](https://github.com/thopiekar/CadIntegrationUtils/blob/master/CommonCLIReader.py) of the file with that plug-in there is no mention of the `objc` module that would be missing. So maybe you have a grossly outdated version? The same bug happens with the CuraBlenderPlugin and the CuraOpenSCADPlugin.
However I don't think these are really causing the issue for you.

And there's another one of those:
```
2021-04-16 09:32:45,518 - ERROR - [MainThread] UM.Logger.logException [112]: Exception: Error loading plugin CuraPowerManagementPlugin:
2021-04-16 09:32:45,532 - ERROR - [MainThread] UM.Logger.logException [116]: Traceback (most recent call last):
2021-04-16 09:32:45,536 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/ultimaker/build/Cura/4.9/build/inst/lib/python3.8/site-packages/UM/PluginRegistry.py", line 443, in loadPlugin
2021-04-16 09:32:45,541 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/Luckys_Trevels/Library/Application Support/cura/4.9/plugins/CuraPowerManagementPlugin/CuraPowerManagementPlugin/__init__.py", line 30, in register
2021-04-16 09:32:45,545 - ERROR - [MainThread] UM.Logger.logException [116]:     from . import PowerManagementOSX as PowerManagement
2021-04-16 09:32:45,550 - ERROR - [MainThread] UM.Logger.logException [116]:   File "/Users/Luckys_Trevels/Library/Application Support/cura/4.9/plugins/CuraPowerManagementPlugin/CuraPowerManagementPlugin/PowerManagementOSX.py", line 20, in <module>
2021-04-16 09:32:45,554 - ERROR - [MainThread] UM.Logger.logException [116]:     import CoreFoundation
2021-04-16 09:32:45,560 - ERROR - [MainThread] UM.Logger.logException [116]: ModuleNotFoundError: No module named 'CoreFoundation'
```
Now this one doesn't have any mirror on Github. It's not on the Marketplace either so I can't see its source code. However since this plug-in is called `CuraPowerManagementPlugin` it could have something to do with how Cura is kept alive for you after it's closed down the main window. Perhaps some thread is still keeping it alive?

thopiekar, 2021-04-20:
Yes. the problem seems to be with CuraPowerManagementPlugin. The problem with many plugins is that there are binary files that depend on Python 3.5, but Cura comes with 3.9, where these binaries are missing now.

At this moment and in the last days, I'm working on a way to build these binaries from source. Before I look them from pypi, but practically I'm was always stuck to the prebuilt binaries offered there and therefore I better take some efforts to build something better now.

That said: All CAD plugins on macOS are dead and all Windows-Automation-based CAD plugins are dead since I depend on pyobjc (macOS) and pywin32 (Windows).

3D-Prints-4U, 2021-04-22:
I removed almost allplugitns and now closes fine thank you 

RON SHANNON
***@***.***
Stay Forever Young
Be Safe

closed 2021-05-04 by Ghostkeeper

Ghostkeeper, 2021-05-04:
We'll close this issue then, as it is more of a bug in the CuraPowerManagementPlugin than in Cura itself.

The problem with Thopiekar's plug-ins was known to us and a major concern for our release, to maybe make the SDK change a major one. That would've automatically disabled the plug-in in Cura 4.9, so it would not have been a problem then. We eventually decided against it after testing the other plug-ins with C libraries linking to Python (e.g. SVGToolpathReader). For them it turned out to not be a problem. Maybe all of them got their libraries from Pipy then. I know I did. I don't know how they build their libraries differently though that it would link to all Python versions correctly. With only Thopiekar's plug-ins having the issue and him already having discovered the issue himself, we decided that it'd be okay to not deprecate the entire contents of the marketplace for this release. That decision had some risk, and that's what we're facing here :(