Ultimaker/Cura #9541: Line Type [Color]: Bridge Mode

KenLucke, 2021-04-08:
### Is your feature request related to a problem?

It is difficult to determine [in the preview pane] where Cura is going to be bridging, using the Bridge Mode settings in Experimental.

### Describe the solution you'd like

It would help tremendously if the bridged areas were shown in another color, as are walls, infill, moves, etc., in the preview pane

### Describe alternatives you've considered

None, really

### Affected users and/or printers

Anyone using Bridge Mode.

rburema, 2021-04-09:
I'll bring it up with the team.

rburema, 2021-04-09:
While we can see the use of a feature such as this, given our current backlog, this is unlikely to be implemented, sorry. We'll have to defer it.

tricky2k, 2021-04-09:
Actually, this goes beyond a new feature a could fix a bug like the described here: https://github.com/Ultimaker/Cura/issues/8778

The fact that Cura doesn't see a bridge as a "bridge", but walls hanging with infill in between, causes the bridge to fail if the orientation of the model doesn't follow certain angles

5axes, 2021-05-02:
Personnaly I can clearly see the benefits ... And this is a basic feature in some other slicer : +1 for me
[image]

tricky2k, 2021-05-02:
This one here is how Prusaslicer sees a bridge. The color is for "bridge", so it's specifically coded in the guts of the software. Wherever there's an overhang or a bridge, the software sees it as a bridge no matter the orientation, and slices it perfectly
https://user-images.githubusercontent.com/74513340/99199139-50919900-279d-11eb-832d-8100b790ef77.JPG

rpodgorny, 2022-02-19:
https://github.com/Ultimaker/Cura/issues/8488

Ghostkeeper, 2022-03-07:
Indeed, this is a duplicate. Please refer to #8488.

closed 2022-03-07 by Ghostkeeper