Ultimaker/Cura #9542: Cura 4.9 beta / preview = freeze or crach computer on Macbook pro M1

ChristopheBis, 2021-04-09:
### Application Version

4.9 beta

### Platform

MacOs big Sur 11.2.3 on Macbook Pro M1

### Printer

Eryone Thinker series

### Reproduction steps

--Similar with all "big" files stl 20Mo and more
Adding object to Cura and slicing (with default profile or custom)

### Actual results

The slicing processing is working but the preview goes up to 99% and stop there.
I must force close Cura or it can also freeze my laptop.
Rarely it's processing with graphic artifact (straight lines) across the screen.
But I'm never able to navigate through the layer.

### Expected results

Should be able to navigate trought the layers

rburema, 2021-04-09:
Just in case: I assume the graphics drivers are up to date as well? With such a new version I'd assume they are, but it does sound like a graphics(-card) related problem.

It would be really helpful to have logs here!

The logs can be found in the Configuration Folder, (cura.log), which can be accessed by 'Help > Configuration Folder'.

ChristopheBis, 2021-04-09:
Yes the laptop is up to date and sorry I was sure the log was attached.

[cura.log](https://github.com/Ultimaker/Cura/files/6285720/cura.log)

The problem is the same with Cura 4.8 and doesn't exist on windows PC with the same Cura backup.
For "huge" files the only work around is to enable "force layer compatibility view" but a lost the navigation as well.

I was hoping it was solved when I read this in the change log
"Various Mac OSX (and other) fixes afforded by upgrades to Python (to 3.8) and Qt (to 5.15).
If you had (UX, visual, graphics card) problems, specifically on (newer) Mac OSX versions, please try this new version."

Thanks

fieldOfView, 2021-04-09:
> I was hoping it was solved

So were we.

ChristopheBis, 2021-04-11:
New log file, now Cura 4.9 is crashing after clicking on preview

[cura.log](https://github.com/Ultimaker/Cura/files/6292311/cura.log)

ChristopheBis, 2021-04-11:
New new log, this time the computer reboot by itself just after slicing and computer the result.

[cura.log](https://github.com/Ultimaker/Cura/files/6292882/cura.log)

rburema, 2021-04-13:
@ChristopheBis If the entire computer is crashing, then this is a bigger problem. No app should be able to do that.

It might still be on ours side, but it also hints that something is wrong on theirs (Mac/OS/graphics drivers) side as well.

Unfortunately, I can't really find anything in those logs right now :-(

colatuttorf, 2021-04-15:
> @ChristopheBis If the entire computer is crashing, then this is a bigger problem. No app should be able to do that.
>
> It might still be on ours side, but it also hints that something is wrong on [...]

I have Cura 4.8 stable crash my entire computer running MacOS multiple times.  It happens reliably when I hit some threshold of number of rotations while viewing the sliced Preview.  I've been able to mitigate the issue by closing Cura frequently.

Definitely a graphics issue that exists within Cura.

Ghostkeeper, 2021-04-16:
Or rather, within Qt. Qt doesn't support the Apple M1 yet. They have an issue open about that, which includes a sub-issue about running via Rosetta: https://bugreports.qt.io/browse/QTBUG-85279
Since Qt is handling the rendering done for Cura, it makes sense that scenes that require a more special rendering (like layer view) could give weird results. There is very little we can do about it until Qt supports your computer.

Villafruela, 2021-04-28:
Could you send us the link to this issue @ Qt, so we can push them to resolve this issue quickly ? thx !

Ghostkeeper, 2021-04-29:
I can't really give you a more specific link than the one I already gave: https://bugreports.qt.io/browse/QTBUG-85279. I don't know exactly which Qt subsystem(s) are causing Cura to crash right now. All I know is that Qt officially doesn't support your computers yet ([source](https://doc.qt.io/qt-5/supported-platforms.html)) and that they still have a bunch of bug reports open, including support for Rosetta2. If you read those comments, it gives the feeling that they themselves also don't know yet which issues are still unreported. So it'll take some time.

Villafruela, 2021-05-15:
thx @Ghostkeeper !

Rose-RJ, 2021-07-27:
I am experiencing the same issue or it does slice 100% but doesn't show the 'preview' I only have this issue with longer prints.. I have tried everything but nothing seems to work.. I work with macOS Big Sur 11.5 and my MacBook is the new 2020 pro with m1 chip.. everything worked fine until the 4.10.. tried to find the older version but can't seem to find it and I have zero clue how to fix the issue.. It's incredibly annoying because I need to add some filament change pauses but I can't.

[cura.log](https://github.com/Ultimaker/Cura/files/6884240/cura.log)
[image]

Ghostkeeper, 2021-08-03:
Apple M1 support is scheduled for Qt 6.2. Unfortunately 6.2 would be a major upgrade for Cura since it drops support for QtQuick Controls 1 which are still used all over the place in our code.

dpkshrma, 2021-11-10:
I am facing the same issue. It seems the [qt issue](https://bugreports.qt.io/browse/QTBUG-85279) was resolved on 25th Nov
@Ghostkeeper is there any update on supporting QT 6.2 in Cura?

nallath, 2021-11-10:
We're still working on it. A major version increase of a library like QT has significant impact on the code, so it's far from a trivial change.

dpkshrma, 2021-11-10:
Thanks for the reply @nallath !

Although, I was able to run the preview by changing a setting `Preferences > General > Tick Force layer view compatibility mode` 

Hope it helps someone until the QT integration is complete.

Cheers!

Ghostkeeper, 2021-11-17:
I'm counting on this being a project of several months, so don't hold your breath yet (not without a hospital nearby anyway).

closed 2024-12-08 by github-actions[bot]