Ultimaker/Cura #9547: Hermera settings, Retraction distance issues

f3bandit, 2021-04-09:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Creality CR-10

### Reproduction steps

Hot-end specs pertaining to finding the cause of issues
E3D Hermera has a max retraction distance of 2mm, beyond 2mm will unload filament from the extruder

Issue 1
Cura not always setting the custom retraction setting in g-code for prints. I had to reset my settings to default before prints each time and then setting each setting again. even when I go thru new project, And importing a new file.
I can go thru profile manager and see the settings are correct, But the retraction setting is not always added to the g-code
of my print.

Issue 2
Most printers are setup to do a purge line before every print. Cura adds the retraction setting after the purge line g-code
which doesn't help at all with the Hermera, As it will use a default retraction of 5mm for the purge line. 
This causes the filament to be completely unloaded before the print actually starts. 

### Actual results

Filament is retracted 5mm and unloaded past the extruder gears.
no way for the printer to push filament thru thru hot-end, But is in
far enough that it doesn't trip my filament sensor.
So I print air forever...

### Expected results

1
If I set a retraction distance of .2mm in my profile it should be added to every print g-code.

2
Retraction distance g-code should be added before any g-code for purge line, as the 
retraction distance should apply to the purge line g-code as well.

[cura g-code fix.txt](https://github.com/Ultimaker/Cura/files/6286442/cura.g-code.fix.txt)

GregValiant, 2021-04-09:
The purge lines come from the printer definition file that was written for Creality by "Trouch.Com" and then submitted to Ultimaker.  You've customized your 3rd party printer with a 4th party hot end/extruder which was installed by a 5th party and your still using the purge lines (created by a 6th party) of a stock printer.

Is that script in the text file your StartUp-Gcode?  Those are hard coded numbers.  If you enter the proper keyword then Cura will insert the appropriate value.  In this case the keyword is "retraction_amount".  The negative sign must remain in place and you must use curly brackets.
G1 F2700 E-{retraction_amount}
Everything affects everything.

f3bandit, 2021-04-09:
yes I have it set for retraction amount, sorry for the wrong term

f3bandit, 2021-04-10:
So this is my start g-code in cura
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G1 Z2.0 F3000 ;Move Z Axis up
G1 F2700 E-0.5

but when I slice and save it stores it like this
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E-15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E-30 ;Draw the second line
G1 Z2.0 F3000 ;Move Z Axis up
G1 F2700 E-0.5

I have checked everything and cura is the only thing with any g-code commands set
also I always print with a brim, and it always stores it in the file to be printed as skirt
it still prints the brim, despite what the file to print states!?!?

GregValiant, 2021-04-10:
It isn't the retraction amount as such.  The extrusion amounts in the first snippet are E15 and E30 as expected, and in the second snippet they are E-15 and E-30.  That will certainly unload your filament (and is really odd and shouldn't happen).

Putting aside the question of the negative sign, your E moves to 15 and then 30 and then to -0.5.  That is a retraction of 30.5.  You should add G92 E0 before the G1 F2700 E-0.5 line.

As for the words ";TYPE:SKIRT" as opposed to ";TYPE:BRIM", that might be because a brim is a skirt printed at a distance of 0 from the model(?)  A question of semantics maybe?

f3bandit, 2021-04-10:
I uninstalled cura and reinstalled cura, then added my printer again from
scratch. so far the issue is gone, no idea why it was doing this!?!?

GregValiant, 2021-04-10:
Double check that the start-gcode is like this:
G1 Z2.0 F3000 ;Move Z Axis up
**G92 E0**
G1 F2700 E-0.5   ;   or E-{retraction_amount}
In a gcode file there will be an M82 before the start-gcode and so the moves are "absolute".

GregValiant, 2021-04-11:
I forgot to answer your question.  I do know why it did that.  The technical term for it is "Stuck on Stupid".  It's something that computers do really well.  People aren't bad at it either.

f3bandit, 2021-04-11:
agreed lol, yeah so far so good, I'm at 67% on a 25 hour print atm. so
fingers crossed :P

Ghostkeeper, 2021-04-20:
So this issue has disappeared for you then?

Cura always copies the start g-code verbatim aside from [filling in the setting keys](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_start_gcode.md) (which is the example with `{retraction_amount}` that GregValiant showed.
I'm unable to reproduce your problem so far. If I fill in your start g-code, that same start g-code pops out. It doesn't suddenly add a `-` sign after the E.

Perhaps there was a post-processing script active? Is it possible to find a project file of what you were using? If not, I think we'll close the issue since you couldn't reproduce it any more either.

closed 2021-05-04 by no-response[bot]