Ultimaker/Cura #9551: Error g-code viewer?

userosos, 2021-04-10:
### Application Version

4.9.0 beta

### Platform

Windows 10 pro 64 bit

### Printer

Ulti

### Reproduction steps

1. Add the model.
2. Multiply the model.
3. Slice

### Actual results

I cansee from ~130 layer that it will printing on first layer. See pic.

### Expected results

I dont want that the hotend goto 1 layer from 130 layer and print.

fieldOfView, 2021-04-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/7412

fvrmr, 2021-04-13:
Hi @userosos thank you for your bug report. 
I can not reproduce this on 4.9 beta with your file, but with another file I can reproduce this issue on 4.9 beta. 
We have a ticket on our backlog for this issue.
Developers see CURA-7960

I will close this issue since it is a duplicate. Thank you @fieldOfView for referring

closed 2021-04-13 by fvrmr