Ultimaker/Cura #9552: The app don't permit editions after change preview mode to prepare mode

andsongomes, 2021-04-10:
### Application Version

4.8.0

### Platform

MacOS X Big Sur

### Printer

Default FFF and Creality3D Ender 3 Plus

### Reproduction steps

1. I put the object on the table;
2. Make the slicer;
3. Show preview;
4. When return to prepare mode, don't permit select any object, is like as disable editing tools.

### Actual results

Don't enable any editing tools or select any object.

### Expected results

Select object to change parameters.

andsongomes, 2021-04-10:
[cura.log](https://github.com/Ultimaker/Cura/files/6290001/cura.log)

[TTSP_cura_Tampa-Pitot-Leandro-v5.3mf.zip](https://github.com/Ultimaker/Cura/files/6290002/TTSP_cura_Tampa-Pitot-Leandro-v5.3mf.zip)

Vandresc, 2021-04-12:
Hi @andsongomes, we fixed this issue in the 4.9 Beta release. Could you please download the build and see if the bug is fixed for you as well? Thank you!

andsongomes, 2021-04-12:
Hi,

Really the bug was fixed on beta 4.9.

Thank you!!!!

____________________
       Andson Gomes

closed 2021-04-12 by Vandresc