Ultimaker/Cura #9554: Cura crashes when using "Save To File"

kmcken, 2021-04-10:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

1. Cura no longer recognizes USB drives for "Save To Drive". It only has the "Save To File" option but I can navigate to the USB drive.
2. It will save early after a PC boot. Hours later, I can slice the .stl file and view the Preview just fine. Press "Save To File".
3. Cura crashes.
4. Bug repeats if Cura is closed and reopened.
5. Bug goes away temporarily after PC restart.

### Actual results

1. No longer recognizes USB drives for Save To Drive option when saving the gcode.
2. Cura crashes when using Save To File.
3. PC restart required.

### Expected results

Save To Drive successfully without crashing Cura.

kmcken, 2021-04-10:
Added log files to the original issue report.

nallath, 2021-04-12:
Duplicate of #9057

kmcken, 2021-04-12:
Can you explain why this is a duplicate? Different behavior occurs doing different things across both issues.

Ghostkeeper, 2021-04-20:
I think that was a mistake. We've been getting a lot of reports about how Cura crashes when ejecting the removable drive.

Ghostkeeper, 2021-04-20:
The log file doesn't mention anything about the crash. That means that the Python environment itself crashed, and that the problem lies within one of Cura's dependencies rather than Cura's own (Python) code.

There is a known issue with the same symptoms as you: https://github.com/Ultimaker/Cura/issues/1537
We've seen this reproduced by installing Windows Explorer extensions, that add things to the open file dialogue, such as Dropbox. Apparently Qt doesn't deal with that well, and it causes Cura to crash when creating the save dialogue. Do you have any such Explorer extensions? And if so, does (temporarily) uninstalling them help to prevent the crash?

In Cura 4.9 we've upgraded the Qt version, so it might also be worth your time to try Cura 4.9. Tell me how that goes!

kmcken, 2021-04-20:
I do not use Internet Explorer and Dropbox isn't linked to my File Explorer. I do have PyCharm Community 2020.2, Python 3.6, 3.8, and Anaconda3 and Anaconda Navigator installed on my PC. None of these were updated between when the Cura 4.8 was working and the bug started.

I have installed Cura 4.9 and it works flawlessly. I like the new features; they are a major improvement over 4.8. However, the Save To Drive option that was in 4.8 doesn't show up when my SD drive is plugged in. In any event, I can save to the SD drive with Save To File with no problems.

kmcken, 2021-04-20:
If I duplicate the crash again, I will upload the new logs.

Ghostkeeper, 2021-04-29:
So in 4.9, for you, this issue transformed into "Cura is not able to find my removable drive any more"?

kmcken, 2021-05-06:
@Ghostkeeper no, sorry that is a reference to my issue #9556.

I just repeated the #9554 crash in 4.9.0 beta.
[cura.log](https://github.com/Ultimaker/Cura/files/6436665/cura.log)

Ghostkeeper, 2021-05-06:
Ah okay. Then we'll close this issue in favour of #9554, to keep the discussions about that crash consolidated.

kmcken, 2021-05-07:
@Ghostkeeper no! Don't close it! I just got this bug in 4.9.0 today and put the log file in the last comment.

Ghostkeeper, 2021-05-17:
With 4.9, we have a similar issue (same result, different reproduction scenario) here: https://github.com/Ultimaker/Cura/issues/9668.

What is different with your case then? The log file looks similar; basically ending abruptly as it seems that Qt has crashed rather than Cura itself. Is there something different about your reproduction steps? Do you have Chitubox' STL thumbnailer installed?

kmcken, 2021-05-20:
FYI - I am currently moving and my PC will be inop through next week meaning I can't test to give you a clearer answer until then.

Ghostkeeper, 2021-06-04:
It would be confirmed by getting an actual stack trace.

Sometimes you'll see more information just by running cura.exe through a command line. All of the things we log should also appear in the cura.log file, but Qt tends to put warnings and errors in stdcout sometimes, which isn't intercepted by Cura to put in its normal log file.

To get an actual stack trace, you'd have to open Cura through GDB or something similar, by:

1. Calling `gdb.exe "C:\Program Files\Ultimaker Cura 4.9\cura.exe"`
2. In GDB, call `run`.
3. As Cura starts, try to trigger the crash by opening a file dialogue.
4. When Cura has crashed, type `bt` in the GDB commands to get a backtrace. This will show the active call stack at the moment the application crashed.

If the most recent entries in the stack are in Qt DLLs, it's a Qt problem. If they are in Python, it's most likely some coding issue in Cura's source code. If they are about cx_freeze, you're looking at the wrong end of the stack.

If you have MS Visual Studio installed, I think the Windows crash dialogue also has an option to "Debug" which would open Visual Studio that also shows a stack trace.

So far we haven't been able to reproduce this issue other than by installing Windows Explorer extensions. Chitubox always crashes, and Dropbox and Google Drive seem to cause a crash only for some people (maybe depending on the version of those extensions).

kmcken, 2021-06-10:
@Ghostkeeper 

I don't use Chitubux, Dropbox, nor Google Drive. I do use OneDrive.

I haven't had this crash occur recently. I think it went away after a windows update. I'll backtrace and upload if I get it again.

closed 2021-07-02 by no-response[bot]

Hughsosexy, 2021-08-14:
[cura.log](https://github.com/Ultimaker/Cura/files/6985804/cura.log)
I am experiencing this issue as well, on all my Curas (4.9, 4.9.1 and 4.10) Even when cura is just open for a few minutes. I cannot work on anything at the moment :(

I have attached the logs

Ghostkeeper, 2021-08-23:
@Hughsosexy That log just ends abruptly, similar to kmcken's log. It indicates that Python crashed. Most likely this is the result of one of the C-libraries we load into Python, with PyQt as the main suspect.

To know precisely what crashed and why we'd need to reproduce the issue ourselves and do a backtrace, perhaps modify the code to skip some things to see where in our code it crashes, that sort of thing. It would help to narrow down what's causing it if you provide information about your computer that could be relevant, e.g. whether you have an STL thumbnail generator installed, if you have any explorer extensions like Google Drive or Dropbox installed, etc.