Ultimaker/Cura #9555: Strange seam behavior

IGO-rek, 2021-04-10:
### Application Version

Arachne

### Platform

Windows 10

### Printer

DeltaBot

### Reproduction steps

1. Opened the file
2. Turned the vase so that the seam was at the back
3. Set parameters: Layer High "0.2", Initial Layer High "0.2", Z Seam Alignment "User Specified", Z Seam Position "Back"
4. I set the value of the nozzle to 0.6 in order to print the vase in two walls
5. Sliced with my custom settings
Preview Mode

### Actual results

I saw that there is a retraction designation on layer 525. I printed a vase, and saw that I have a strange line almost along the entire length where there was a point showing a retraction.

### Expected results

I have been using Сura since 15.04.6 version. But most likely I still don't understand much, for me now it is important to make sure that the seam is aligned along the entire length and does not spoil the appearance of the vase.

IGO-rek, 2021-04-10:
[image]
[image]
[image]
[image]
[image]

IGO-rek, 2021-04-10:
[image]
[image]

https://user-images.githubusercontent.com/82113847/114277939-32c81780-9a4f-11eb-924f-d8ccc4b7e0c4.mp4

IGO-rek, 2021-04-10:
[Strange seam behavior.zip](https://github.com/Ultimaker/Cura/files/6290314/Strange.seam.behavior.zip)

IGO-rek, 2021-04-10:
[Strange seam behavior.zip](https://github.com/Ultimaker/Cura/files/6290324/Strange.seam.behavior.zip)

fvrmr, 2021-04-13:
Hi @IGO-rek thank you for your bug report.
I can reproduce this issue in Arachne, I will discuss it with the team. 

But in 4.9 beta the seam looks better and is better aligned. 
[image]

Can you let me know if you can print better with 4.9 beta?

IGO-rek, 2021-04-20:
> Hi @IGO-rek thank you for your bug report.
> I can reproduce this issue in Arachne, I will discuss it with the team.
> 
> But in 4.9 beta the seam looks better and is better aligned.
> [image]
> 
> Can you let me know if you can print better with 4.9 beta?

Hi @fvrmr, yes this version is much more stable, I will use it for now. During printing, my flash drive on which there was a code was disconnected, so part of the vase was not printed. But I think that that part is enough to analyze. Those large gaps on the seam indicate too much retraction, I need to adjust it myself. Almost all new layers start from the same point, so I am happy with using cura 4.9, but not Arachne.
[image]
[image]
[image]
[image]
[image]

fvrmr, 2021-04-29:
Hi @IGO-rek thank you for letting me know that is works for you. 
We have a ticket on our backlog for the seam in Arachne.

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?  (The Z-seam has undergone a lot of work recently.)

closed 2024-11-23 by IGO-rek