Ultimaker/Cura #9556: "Save To Drive" option no longer available

kmcken, 2021-04-10:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

Issue #9554 started occurring at the same time.

1. Using Cura through my usual workflow.
2. "Save To Drive" option is no longer available regardless of the presence of USB drives or not.

### Actual results

"Save To Drive" option is no longer available regardless of the presence of USB drives or not.

### Expected results

"Save To Drive" option should be available when USB drive recognized by PC.

kmcken, 2021-04-10:
Issue #9554 and #9556 began occurring simultaneously. A Reddit user (not me) posted about this back several months ago.

fvrmr, 2021-04-13:
Hi @kmcken thank you for your bug report. 
Looking at your log files it looks like there is some problems with permissions on your computer.
```
2021-04-07 13:18:04,645 - DEBUG - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [85]: Writing to G:/CE5_MenMagV1LoadingTool.gcode
2021-04-07 13:18:04,648 - ERROR - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [105]: Permission denied when trying to write to G:/CE5_MenMagV1LoadingTool.gcode: [Errno 13] Permission denied: 'G:/CE5_MenMagV1LoadingTool.gcode'
```

Could you try if this happens on 4.9 beta?

kmcken, 2021-04-13:
This is strange because it used to work before. I haven't made any changes to the PC. Is there a setting on Win10 to give Cura permission?

I'll try the 4.9 beta.

fvrmr, 2021-04-15:
Yeah it is indeed strange, because before here I can see in the logs that it is working. Let me know if it works on 4.9 beta.

kmcken, 2021-04-20:
I have installed 4.9 and Save To Drive isn't available when my SD drive is plugged in to my USB reader.

Attached is my log from 4.9. FYI - I use Dark Mode.

[cura.log](https://github.com/Ultimaker/Cura/files/6345299/cura.log)

Ghostkeeper, 2021-05-03:
Perhaps an antivirus software is preventing access to removable drives?

So far I've not been able to reproduce it. But if Windows tells Cura that it doesn't have access, but then allows other programs to access it, I can't think of anything else than access control blocking it.

kmcken, 2021-05-06:
@Ghostkeeper It's weird because I can still save to removable drives through File -> Export and "Save to Disk"

kmcken, 2021-05-06:
@Ghostkeeper I wonder if since the SD reader is a USB adapter, if after a restart Windows or Cura no longer considers it a removable drive?

Ghostkeeper, 2021-05-07:
Cura finds the removable drives on Windows by:

* Asking Windows for which drive letters are logical drives (`GetLogicalDrives` call).
* Erasing drive letters A and B from those. Cura will never detect drives A and B.
* For every logical drive: Ask Windows if the drive type is a removable drive ([using the GetDriveTypeA system call](https://github.com/Ultimaker/Cura/blob/master/plugins/RemovableDriveOutputDevice/WindowsRemovableDrivePlugin.py#L66)).
* If the file system of the drive returns an empty string, skip that drive.
* If the drive has 0 bytes of free space, skip that drive (skips over empty SD card readers).

So perhaps the drive is full? Or maybe it can't read the file system and free space?

kmcken, 2021-06-10:
@Ghostkeeper 

My Cura 4.9.0 instance now recognizes removable drives. I have done nothing and now suspect a windows update.

fvrmr, 2021-06-14:
@kmcken so now the save to removable drive is working for you?

kmcken, 2021-06-14:
@fvrmr correct and I have no idea why

fvrmr, 2021-06-15:
I will close this than as solved. 
You can always reopen this or report new bugs.

closed 2021-06-15 by fvrmr