Ultimaker/Cura #9557: Selecting a surface to lay flat uses the wrong surface when the object is mirrored

dwery, 2021-04-10:
### Application Version

4.9.0-beta

### Platform

Linux

### Printer

Ender 3

### Reproduction steps

Mirror an object
Select a surface to lay flat

### Actual results

Object lies on the opposite surface

### Expected results

Object should lie on the selected surface

fieldOfView, 2021-04-11:
As a workaround, you could use the "Apply transformations to mesh" function provided by MeshTools.

fvrmr, 2021-04-16:
Hi @dwery could you let us know if the suggestion of @fieldOfView works for you.

dwery, 2021-04-16:
I would rather not use a plug-in to workaround a bug

fieldOfView, 2021-04-16:
In other words: you would rather have something that does not work than use a workaround?

dwery, 2021-04-16:
As a workaround I can simply click on the opposite surface

Ghostkeeper, 2021-04-20:
I can reproduce this. This was discovered in December 2019 as well (reported by me in Ultimaker's bug tracker), but then the testers couldn't reproduce it any more. I'll report it again to see if people can reproduce it now.

Ghostkeeper, 2021-04-21:
Developers, see issue CURA-8183.