Ultimaker/Cura #9558: Coasting before tool changes

tommylux, 2021-04-11:
### Is your feature request related to a problem?

On an multi extruder printer, when using custom tool change scripts to perform purging for example,  there is often stringing despite retraction settings are applied. When inspecting the moves, you can see that coasting is not applied before tool changes. 

### Describe the solution you'd like

When 'Enable coasting' is enabled, I would like cura to coast before tool changes.

### Describe alternatives you've considered

I have tried increasing retraction settings, reducing print speed. However, because this printer uses a shared nozzle, it is difficult to reduce pressure without any of the features offered by coasting.

### Affected users and/or printers

A20T.

fvrmr, 2021-04-13:
Hi @tommylux thank you for your feature request.
Can you explain to me what kind of tool change scripts you are using? 

If I understand you correctly you want coasting before the material changes right?

tommylux, 2021-04-13:
Certainly, the following depicts the code inserted before and after T{} commands.

;Change End (happens first)
	G10 ; Retract
	G91 ; Relative position
	G0 Z0.2 F9000 ; Lift Nozzle
	G0 Z1 X-5 F9000 ;lift diagonal

	G90 ; Absolute
	G0 X0 F6000; Go to bucket 
	M211 S0 ; Disable SoftStops
	G0 X-5 F500; Go to bucket Closer
	G11 ; Recover
;/Change End

;Change Tool
	M109 R{material_print_temperature}
	G91 ; Relative
	G1 E25 F100 ; Extrude into Bucket
	G1 E5 F50 ; Slow Purge to reduce pressure
	G92 E0 ; Set Extruder Position
	G90 ; Absolute
	G10 ; Retract
	
	; Wait for ooze
	G4 S3 ; Wait 2 seconds

	G0 X5 F500 ; Wipe
	G0 X-5 F4000 ; Wipe
	G0 X5 F500 ; Wipe
	G0 X-5 F4000 ; Wipe
	M211 S1 ; Enable SoftStops
;/Change Tool

[image]
When changing tools, coasting is not performed, (even when using a purge tower) so when the tool change scripts are applied, there has not been an opportunity to reduce pressure in the nozzle and stringing occurs.

As a possible work-around, I'm considering adding a poor mans coasting (1mm circle) at the current position on tool change.

Ghostkeeper, 2021-04-28:
Your problem is not that coasting is not being applied before tool changes. Your problem is that it's not being applied to skin or infill. Indeed, Coasting only applies to contours. It's considered part of the closing of a seam. As such, it's not expected to apply to skin or infill, which do not have seams, unless you use the concentric pattern for those.

This is currently expected behaviour. Going by the setting description though, it might not be what you want to use Coasting for.

tommylux, 2021-04-28:
Perhaps a feature request could be to add coasting to infill or skin. Whichever way you skin the cat, the ideal solution would be to coast before changing tools!

Ghostkeeper, 2021-04-28:
Try changing the top/bottom pattern to Concentric and observe how coasting works fine then. It's about the pattern being used.

tommylux, 2021-04-28:
No sign of actual coasting when changing the settings, but there are extra travel moves, though I am not sure of the concentric patten.
[Image.jpeg]

fvrmr, 2021-04-30:
Coasting with concentric is working for me. 
[image]
I can't see your screenshot because you've replied by email and then Github doesn't attach the image.

tommylux, 2021-04-30:
Are you using two colours in your model? I’ll send new screenshot shortly. Bear in mind, right before the tool change, the gcode is extruding.

[Screenshot (157)]<https://user-images.githubusercontent.com/77000924/116689055-e28e0680-a9b7-11eb-9ea6-d92fa088ee22.png>

Ghostkeeper, 2021-05-03:
This problem has been exacerbated in Arachne because many walls are no longer closed loops. It'll need to be fixed before then. Developers, see issue CURA-8207.