Ultimaker/Cura #9560: Bed temperature not pulled from material settings

npburns224, 2021-04-11:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Elegoo Neptune 2

### Reproduction steps

1. Select custom filament from material list
2. Observe that the material's printing temperature is applied, but the build plate temperature remains the same.

### Actual results

Only the printing temperature is updated with the value entered in the selected material's print settings.

### Expected results

Both the printing temperature and the build plate temperature should have been updated with the values entered in the selected material's print settings.

fvrmr, 2021-04-12:
Hi @npburns224 thank you for your report.
Could you share your material profile? I can't reproduce it with mine profiles.

npburns224, 2021-04-12:
Sure, I've attached my profile for the PLA and PETG I'm using.
[Material Profiles.zip](https://github.com/Ultimaker/Cura/files/6300263/Material.Profiles.zip)

ksihota, 2021-04-15:
You might try this to see if it applies...
If you are using a profile that was derived from one that had the material temps set to a specific number then your new profile will probably retain the number rather than the temps from the materials profile. The settings are defined in the quality_changed folder in the matching cfg file. There can be a lot of these so make sure you change the proper one. 
It looks something like this:
material_print_temperature = 195.0
You can remove the line or change it to something like this:
material_print_temperature = =default_material_print_temperature
same applies to the material_bed_temperature

Be very careful when changing these cfg files as they are made and changed by the program so making changes to these files can break your profiles or cause other unexpected results. You can always make a copy of the file so you can test it and then switch it back if something breaks.

It would be nice to have something in the UI that would allow the user to revert this for any profile back to the defaults provided by the material profiles but this might cause issues for some, depending on their work flow?

npburns224, 2021-04-17:
After removing all references to `material_bed_temperature` within the default profile I'm using, Cura now just shows a 0 for the Build Plate Temperature.

The value still doesn't update based on what I've entered in my Materials settings. It's strange because every other setting I've entered appears to update when I change to another material, plus I'm using the default profiles as they were installed.

ksihota, 2021-04-18:
What file did you alter? There should only be one instance for the specific profile change. If it is indicating a 0 then I would add the line
material_bed_temperature = =default_material_bed_temperature
note the extra =
or switch the file back and put another number in place of what was there originally and see if the new number displays when you start Cura again.
Again, if you are not in the proper file it will not work. I have changed the assigned number to something different that I would recognize in order to verify I am using the right file. If it doesn't display the new number I change the number back and check for the proper file again. Don't change a bunch of things unless you know for sure what you are doing. make one change at a time and revert the file back if it is not right or you will end up with a mess.
Keep in mind that in some cases you have to restart Cura in order to see changes. I can't remember if this is one of the times. I sometimes get around this by switching between a couple of different printers so Cura will reload the necessary files.

ksihota, 2021-04-18:
I believe the file is an extruder cfg file found in quality_changes
it might look something like this
predator_extruder_0_%232_petg_0.3.inst.cfg

Ghostkeeper, 2021-04-28:
Based on the results shown in this thread so far, and that the Elegoo Neptune must be a custom definition file, I think we can conclude that this is indeed just a mistake in the profiles.

The temperature in the material profile is set as the Default Printing Temperature. If the default formula for the Printing Temperature was modified so that it no longer depends on the Default Printing Temperature (e.g. because the user overrides it to a fixed value, or because you remove any references to `default_material_bed_temperature` then indeed it should no longer adjust based on the temperature set in the material profile.

closed 2021-04-28 by Ghostkeeper