Ultimaker/Cura #9561: Turn On/Off 3D printr model around bed

Zogar89, 2021-04-11:
### Is your feature request related to a problem?

The printer model around the bed can be annoying or remove performance in some cases.

### Describe the solution you'd like

An option in preferences to enable or disable the printer model.

### Describe alternatives you've considered

Do nothing.

### Affected users and/or printers

All. Low Specs PC.

fvrmr, 2021-04-13:
Hi @Zogar89 thank you for your feature request.
There is a work around and that is remove the file of your printer here: `C:\Program Files\cura version\resources\meshes`

But I will discuss this with the team. Keep you posted!

fvrmr, 2021-04-16:
Hi @Zogar89 we have discussed it with the team and we decided to reject this feature request. Because not many users will use this and you can make it work with the work around I mentioned above.  
Hope you understand.

closed 2021-04-28 by Ghostkeeper