Ultimaker/Cura #9562: Missing section if stl is only printed once

M-Krieger, 2021-04-11:
### Application Version

4.8.0

### Platform

MacOS 10.14.6

### Printer

HevoRS

### Reproduction steps

Hey there.

I had a request to print a stl file and had issues with missing section until i realized they aren't missing if I multiply the stl. A simple slice with an up-to-date Prusaslic3er had no issues.

Here is the stl which I used:

### Missing sections with default settings
* open the stl
* slice it with my default settings
* **Missing sections**
[image]

### Missing sections with multiplied model but print one at a time
* multiply model on print area
* slice again
* set print sequence to "One at a time"
* *Missing sections*
[image]

### No missing sections with multiplied model but print all at once
* multiply model on print area
* slice again
* set print sequence to "All at once"
* *No missing sections*
[image]

### Actual results

There are missing sections between walls in the upper part of the model

### Expected results

I expect the mentioned sections to be solid.

fieldOfView, 2021-04-11:
Please include a link to a project file (File -> Save Project...) so we can have a look at the model and your settings.

M-Krieger, 2021-04-11:
Thank you for your reply. I added it as a zip file because github won't let me upload .3mf files.

[CFFFP_Aufsatz v4 v2.3mf.zip](https://github.com/Ultimaker/Cura/files/6291861/CFFFP_Aufsatz.v4.v2.3mf.zip)

fvrmr, 2021-04-16:
Hi @M-Krieger thank you for your bug report. You could try to enable the setting `union overlapping volumes`. 
Let me know if that solves your problem!

Ghostkeeper, 2021-04-29:
These are indeed self-intersecting meshes. The purpose of Union Overlapping Volumes is to choose whether the overlap of two volumes should be considered outside of the model (convention in some geometry systems) or considered to be inside of the model (more robust for 3D printing, and for meshes that were produced for rendering). So this is expected behaviour.

closed 2021-04-29 by Ghostkeeper