Ultimaker/Cura #9565: Multiply custom supports

Zogar89, 2021-04-11:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

All

### Reproduction steps

1-Do a custom support.
2-Right click, multiply
3-The new object is positioned outside of the build plate and is not visible.

### Actual results

The multiplied support is not visible.

### Expected results

We should have a new support.

fvrmr, 2021-04-12:
Hi @Zogar89 thank you for your report.
I am not able to reproduce this. Does it happen with every model you've selected as support?

Zogar89, 2021-04-12:
[Multiplicate supports.zip](https://github.com/Ultimaker/Cura/files/6300250/Multiplicate.supports.zip)

Very simple. Try that.

https://youtu.be/HBNwZiiN0yk

fvrmr, 2021-04-16:
Hi @Zogar89 it seems like this has to do with your plugin for custom support. Could you try if this also happens with the `mesh type: print as support`?
If the problem is with the plugin you need to contact the plugin developer. Since we don't maintain the plugins.

Ghostkeeper, 2021-04-29:
Couldn't reproduce this issue either with the Custom Support plug-in (as used in the video) nor with the support blocker changed to a support mesh in the per-object settings tool.

Zogar89, 2021-04-29:
@Ghostkeeper 
> Couldn't reproduce this issue either with the Custom Support plug-in (as used in the video) nor with the support blocker changed to a support mesh in the per-object settings tool.

Note:In my video the drop-down menus are not registered with the right click.
I right click multiply object.

With the support blocker in overlay modifier mode I can't multiply the object. After clicking on the box that asks about the amount to multiply, nothing happens.
Nor I can multiply blockers supports. 

If the blocker supports is grouped with an object multiplied.

Ghostkeeper, 2021-04-29:
Yeah I figured that the context menu didn't show up in the video (or that you used the Ctrl+M hotkey), and also the pop-up asking how many copies you want. But yes, your reproduction steps are exactly the same as mine then. Yet I can't reproduce the problem.

Could you provide us with a log file? If anything went wrong, it should appear in there.

Zogar89, 2021-05-13:
This issue has apparently been read by the plugin creator and has been resolved.

Thanks

closed 2021-05-13 by Zogar89