Ultimaker/Cura #9566: Cura behavior creates many starting points that decrease quality of print

pierremilcent, 2021-04-11:
### Application Version

4.8.0, 4.9.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

Using Surface mode in the joined profile with this STL part creates the result shown in the screenshots.

### Actual results

All the starting point are resulting in a bad print

[image]

### Expected results

This is the projet used.
I expect to have a smooth surface where starting points are gathered rather then very bad result.
I am using direct drive with slow speed (1.5mm at 35mm/s, printing 30mm/s)
[E3PROSKR_Cowling_1_profile3_bb_PPPPLA-3-43-025.zip](https://github.com/Ultimaker/Cura/files/6292901/E3PROSKR_Cowling_1_profile3_bb_PPPPLA-3-43-025.zip)

fvrmr, 2021-04-12:
Hi @pierremilcent thank you for your report. 
It looks like your model is not manifold. And I think that might be the issue of creating a lot of starting points. 
Is this your model? Because then you could try to fix it and see if that solves the problem.

pierremilcent, 2021-04-12:
Hello @fvrmr 
thanks for your answer. You may be right indeed, Cura says the model should be fixed.
But this design is made by a company who knows Cura and its capabilities. By using the "surface" mode and deactivating "Union volumes overlapping" they are able to print very nice parts at light weight : plane print https://www.planeprint.com/print
My question is so to understand why Cura is creating such starting points exactly at this place. Whatever options i am using, tryingo move the z seam has no effect, only this side of the part is affected ? The part is symertric why not the other side.
This is my question and how can you look at this ? 

Thanks.

GregValiant, 2021-04-12:
"...is not watertight and consists of 101 sub-meshes".  Each sub-mesh is separate and is treated separately.  Each has it's own starts and stops.  Each has it's own Z seam.  After exporting the model from Cura as an STL and having it repaired at Netfabb it came back and sliced as seen below.  The repaired model is on the left.  The top inside edge is very different.

[image]

pierremilcent, 2021-04-12:
Great @GregValiant !!! Any way to do the repait myself or having access to the modified STL ? 
I also find interesting to mention this to the designer in order to improve the original piece if possible.

Thanks for your advice.

GregValiant, 2021-04-12:
It's an AutoDesk site at [Service.Netfabb.com](https://service.netfabb.com/).  You have to create a free account.  You upload a file and a few minutes later download the repaired file.  There have only been two that I found couldn't be fixed.

pierremilcent, 2021-04-12:
Thanks @GregValiant i did not know this eervice and it seems to do the job.
But the process changed quite a lot some parts of the piece. As you can see below on the left this is the remix part and on the right the ortiginal.
Some area have been removed, which avoids now to have this bad print with many starting point. 
I wonder how the part will be now assembled since these areas are aimed to insert join with the other side part ...
I need to print it and check how to solve this situation but it could work.

[image]

GregValiant, 2021-04-13:
I think you need to go back to the designer and point out the problems you are having.  I noticed that some things had changed after repair.  A single model comprised of 101 separate meshes isn't good.  The un-repaired file would probably print but end up pretty ugly.

pierremilcent, 2021-04-13:
Thanks @GregValiant i sent an email to the designer explaining what you helped me to found.

pierremilcent, 2021-04-13:
Thanks to all who answered. With your help and request to the designer i have now a new part to print which could produce a better result as you can see below with far less starting points and a well positionned seam on the side.

[image]

GregValiant, 2021-04-13:
Thin walled models are often a problem.  They need to be examined closely and generally they can't be scaled down because features start to disappear as they get too small to print.  Congrats.

fvrmr, 2021-04-13:
Nice to see that the model is fixed and the seam is placed correctly now! Let me know if you have more questions.

I will close this issue since it is solved!

closed 2021-04-13 by fvrmr