Ultimaker/Cura #9567: Will not slice any object

suckatprinting, 2021-04-12:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

No matter what i do, cura will not slice any object. the slice bar just stays white indefinitely and there is no process in the background. I have allowed the app all permissions possible and I can't find a fix on the internet anywhere.

### Actual results

I cant fix it

### Expected results

Hopefully someone smarter than I am will have a better idea

nallath, 2021-04-12:
Please attach the Cura log files as indicated by the bug template.

pierremilcent, 2021-04-12:
Did you try to move the z seam by it location X and Y.
I found that depending on z seam location sometimes the slicing process if blocked, changing value of X & Y solves this, then you have to find the right place.

nallath, 2021-04-12:
> Did you try to move the z seam by it location X and Y.
> I found that depending on z seam location sometimes the slicing process if blocked, changing value of X & Y solves this, then you have to find the right place.

Could you share a project where it's blocked? It shouldn't be blocking based on the z-seam position.

GregValiant, 2021-04-16:
On the one hand it sliced fine either way.  On the other hand...

"...is not watertight and consists of 4 sub-meshes".
[image]

I noted this in the sliced model.  I don't know if it's intended or not.  Structurally it doesn't look too good.

[image]

pierremilcent, 2021-04-16:
The part is designed for such purpose and print is going well at the end with very good quality?
The fact is that depending of seam position the slicing process may be "blocked", never finishing, or it does work.

I think it is true for objects with no default.

closed 2021-04-26 by no-response[bot]