Ultimaker/Cura #9570: Version 4.9 removed the thin-walled horizontal fill

coolgua888, 2021-04-12:
### Is your feature request related to a problem?

1.I used 4.8 slices, a shell just 1mm thick, which was divided into an outer wall of 0.4 and an inner wall of 0.4. The sawtooth or mesh filling in the middle could get the slices as shown in the figure, which could have good strength without delamination.
[image]

[image]

2.However, I used 4.9 cura slices, and the middle was directly transformed into a straight line. Without transverse filling, the strength would become very poor and the layer would rise.The Settings of the two softwares are the same, hoping to restore to the level of 4.8.
[image]
[walls.zip](https://github.com/Ultimaker/Cura/files/6298111/walls.zip)

### Describe the solution you'd like

4.9 Removed horizontal filling of thin wall, thin wall strength will be reduced, will be stratified

Thin wall with transverse filling, firm

### Describe alternatives you've considered

4.9 Removed horizontal filling of thin wall, thin wall strength will be reduced, will be stratified

Thin wall with transverse filling, firm

### Affected users and/or printers

all

fvrmr, 2021-04-13:
Hi @coolgua888 thank you for your feature request.
Could you share a project file (file - save project) instead of a stl. In that way it is easier to see the difference and reproduce. 

Thank you in advance.

closed 2021-04-27 by no-response[bot]

Ghostkeeper, 2021-04-30:
It's not expected to overextrude the infill pattern on top of gap filling. The second picture was a bug.