Ultimaker/Cura #9571: unnecessary, time consuming traveling in top-layer

jon4sreiter, 2021-04-12:
### Application Version

4.8.0

### Platform

Windows 10 (64 bit)

### Printer

Creality CR-10S Pro

### Reproduction steps

1. loading file "File_input.3MF" into Cura
2. Slicing file (with the settings provided by the project-file "Savefile_cura.3mf")
3. Saving gcode to sd card
4. printing file

### Actual results

A lot of unnecessary traveling while filling the top-layer-parts of layers 114 to 116 as shown in the pictures (g-code provided in file "Haube Lufternabe.gcode") resulting in very long print times of the affected layers.

### Expected results

Filling the top-layer-part of the affected layers without traveling large distances after just a few mm of printing.

jon4sreiter, 2021-04-12:
I think i made a mistake uploading the files.
So here they are (i hope):

https://user-images.githubusercontent.com/82399751/114434833-c0c50f00-9bc3-11eb-9735-2326e8c1dfce.mp4

https://user-images.githubusercontent.com/82399751/114434848-c4f12c80-9bc3-11eb-8d17-13f89122e8bb.mp4

[image]
[3mf_files.zip](https://github.com/Ultimaker/Cura/files/6298582/3mf_files.zip)
[g-code_file.zip](https://github.com/Ultimaker/Cura/files/6298583/g-code_file.zip)
[image]
[image]

GregValiant, 2021-04-13:
When angles on a part are very close to the Line Direction of skin, infill, etc., it does make for a lot of movement when combing is enabled.
The default line directions are [45,135] which can be expressed by empty brackets [].  You can change them to whatever you want like [70,100] or [22,44,66] or whatever is necessary to get rid of the very slightly angled lines you see near the nozzle in your image.  Adjusting the combing mode can also help.  It's unfortunate that Combing is not an option for a Support Blocker.

fvrmr, 2021-04-13:
Hi @jon4sreiter thank you for your report.
And thanks @GregValiant for your answer. I was just typing that this has to do with combing mode. 😄 

Let us know if the suggestion above works for you.

jon4sreiter, 2021-04-13:
Thank you for your answers @GregValiant  and @fvrmr .
I tried changing both the _Top/Bottom Line Directions_ and the _Infill Line Directions_ (also both at the same time) with @GregValiant 's proposed values and also with some random ones, but the resaults are much or less the same:
[image]
[image]
[image]
[image]

I recognized that all the travel lines are going though the same two corners in the left side of the part, independently of the chosen directions.
Changing the _Combing Mode_ from _Not in Skin_ to _Within Infill_ removed the problem:
[image]

GregValiant, 2021-04-13:
I'm sorry to have confused you.  I just threw those numbers in as an example of what you could do.

When skin lines and walls are _almost_ parallel and combing is on, then there is a lot of travel.  Another alternative would be to use the Concentric pattern for tops and bottoms.  By their nature they aren't as strong because each layer can be identical to the one below.

A septagon eh?  Isn't that one of the Transformers?

Ghostkeeper, 2021-04-30:
This problem is secretly a duplicate of https://github.com/Ultimaker/Cura/issues/4157.

The travel moves are inefficient because the ordering of the skin lines is determined in a nearest-neighbour fashion. This means that at the end of every skin line, it's going to look at which other skin line (and endpoint) is closest. It'll continue with that line. This is pretty naive but in most cases produces decent results.

In the case of this layer, the next nearest line endpoint happens to always be on the other side, due to the thin strip of skin and the skin lines being angled almost parallel with the strip:
[image]

Because of the combing setting, it's not allowed to cross the skin. It should go around the skin if possible. And in this case, it is possible. Near the screw hole there is a gap in the skin so all of the travel moves can go through there to avoid going over the skin.
[image]

If you change the combing mode, then either it's allowed to go through the skin ("All"), or it's not allowed to go through the skin or the walls ("Infill Only"). In the latter case, there is no combing path possible and so instead it's supposed to make a retraction and then travel directly without combing. The combing requires that you'd go around the skin there, and the combing works correctly.

The real problem is then that the nearest-neighbour optimisation looks at the distance from one line segment to all the other line segments, but the distance is not a good approximation in this case for how much time it will take to travel there, because it's not travelling in a straight line. It may be the closest endpoint, but because the endpoint is on the other side of a very long strip of skin it'll take a long time to get there.

Instead, if the nearest-neighbour calculation would take the distance of the combing path into account in this calculation, it would find that the travel move to the opposite side is longer than the travel move to the next skin line on the current side, and so it would stay on the same side of the strip of skin and not make these huge combing moves. The problem here is performance though. There are quite often several hundred skin lines in a part, and you'd have to calculate the combing path for each of these. Calculating the combing path means calculating intersections with all of the areas you need to avoid, for every segment of the combing path. To do that for all combinations of skin lines means the algorithm becomes O(N⁴) which grows very fast with the resolution/size of the model. That is why [we implemented this for infill](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/infill_enable_travel_optimization.md) but not for skin. The infill normally has way fewer (poly)lines to optimise. We experimentally found that if the optimisation was also applied for skin, it would on average increase the slicing time by ~25% without any reduction in travel time. And in extreme cases it increases the slicing time by a factor of 30.

So to get a good implementation of #4157, we need something better. I'm thinking along the lines of first finding several of the nearest line segments, then looking for the nearest one how long the combing path is. If this combing path is smaller than the distance to the second-nearest endpoint, it's fine and the nearest endpoint is chosen. Otherwise, it'll need to look at the combing path of the second-nearest endpoint, and so on. This could take the computation time back down from O(N⁴) to O(N²).

closed 2021-04-30 by Ghostkeeper

jon4sreiter, 2021-04-30:
@Ghostkeeper thank you for your very detailed explanation! 😃
I've never had this issue before and it only affected three layers on this part. I think it's not that big of a deal.

@GregValiant No, it has nothing to do with the Transformers, it's some kind of cover for an industrial fan. 😁
[image]