Ultimaker/Cura #9572: Print is at 308% and rising

Blake-McCullough, 2021-04-13:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Ender 3 pro

### Reproduction steps

I set my printer to print to my ender 3 pro (with an mks sgen lv2 board) via usb, and i take a dumb, return and see it at 308%, yet still printing and rising...

### Actual results

i got 308%

### Expected results

it should only be 8%

GregValiant, 2021-04-13:
Are you using a Post-Processor like "Display Progress on LCD"?  It uses M73 to set the percent on the LCD.  That isn't enabled in the firmware on my Ender 3 Pro in which case the Marlin firmware itself handles the % complete as "BytesProcessed / FileSizeInBytes".

Cura itself doesn't have anything to do with the % as displayed on the LCD but that Post Processor might.

Ghostkeeper, 2021-04-16:
Maybe it's like, based on the time estimates?

closed 2021-04-27 by no-response[bot]